jiangdongguo/AndroidUSBCamera #218: How to convert frames' byte[] (provided by setFrameCallback) to org.webrtc JavaI420Buffer?

acrolink, 2020-04-07:


konovalov-k, 2020-04-08:
Hi again @acrolink, just use as 
`private val myFrameCallBack: IFrameCallback = IFrameCallback { frame ->
        val videoFrame = VideoFrame(
            NV12Buffer(
                width,
                height,
                width,
                height,
                frame,
                { JniCommon.nativeFreeByteBuffer(frame) }), 0, System.nanoTime() - start
        )
        capturerObserver.onFrameCaptured(videoFrame)
    }
    camera.setFrameCallback(myFrameCallBack, UVCCamera.PIXEL_FORMAT_NV21)  
`

konovalov-k, 2020-04-08:
https://gist.github.com/konovalov-k/a8e834465fc26e7fd67c96d6b3af1513

acrolink, 2020-04-08:
@acrolink 

Thank you. I have managed to make it work on the rate of 95%. The picture is fine but sometimes the image gets green or distorted.

What is working for me:
`camera.setFrameCallback(mIFrameCallback, UVCCamera.PIXEL_FORMAT_YUV420SP);`
and:
```
public void addFrame(ByteBuffer frame) {
 
            byte[] imageArray = new byte[frame.remaining()];
            frame.get(imageArray);
            frame.rewind();

            NV21Buffer nv21Buffer = new NV21Buffer(imageArray,640,480, null);
            VideoFrame videoFrame = new VideoFrame( nv21Buffer, 0, System.nanoTime() );
            capturerObs.onFrameCaptured(videoFrame);
}
```
Correct me if I am doing something wrong.

konovalov-k, 2020-04-08:
About distorted / green image... I don't know why. I got the same problem.
Is PIXEL_FORMAT_YUV420SP work well with NV21 buffer? I had some problems in past with that stuff. 
Also why you don't provide lambda for clear buffer on native side?

acrolink, 2020-04-08:
Could you please give an example of using a lambda in this context.

konovalov-k, 2020-04-08:
As I did here NV12Buffer( width, height, width, height, frame, { JniCommon.nativeFreeByteBuffer(frame) }), 0, System.nanoTime() - start )

acrolink, 2020-04-08:
I have changed my code to this:

```
  public void addFrame(ByteBuffer frame) {
            long start = System.nanoTime();
            byte[] imageArray = new byte[frame.remaining()];
            frame.get(imageArray);
            NV21Buffer nv21Buffer = new NV21Buffer( imageArray, 640, 480, () -> JniCommon.nativeFreeByteBuffer(frame));
            VideoFrame videoFrame = new VideoFrame( nv21Buffer, 0, System.nanoTime() );
            capturerObs.onFrameCaptured(videoFrame);
}
```
Now the issue of green and corrupted frames happens less frequently but I guess to eliminate it completely I may have to validate the `videoFrame` as valid (good) before sending it over to `capturerObs.onFrameCaptured()`. Any ideas regarding this?

konovalov-k, 2020-04-09:
Maybe get bytes from U and V channel, mix it and if in random places it green don't send?)

konovalov-k, 2020-04-09:
@acrolink And i think it is not the issue of this lib :)

acrolink, 2020-04-10:
> Maybe get bytes from U and V channel, mix it and if in random places it green don't send?)

Could you kindly refer me to some tutorial online on how to do that?

konovalov-k, 2020-04-10:
It is just idea based on colour theory. I don't have tutorial or solution too. :(

acrolink, 2020-04-12:
I suspect that the corrupted and green frames issue is related to WebRTC's `VideoFrame` itself since the same frames provided by UVC's `iframeCallback` when passed to OpenTok's or Agora's API produce perfect video play.

konovalov-k, 2020-04-13:
It may be wrong, because i use this lib in other cases and still getting green images from time to time.

pumatus, 2020-06-22:
@konovalov-k Hi~ Is this problem resolved?

konovalov-k, 2020-06-22:
Hi @pumatus. Main problem with convertation or with green frames from time to time?

pumatus, 2020-06-22:
@konovalov-k  Hello~ I want to send the uvc preview data (byte[]) to webrtc, there is a problem here, how do you deal with it?

konovalov-k, 2020-06-22:
@pumatus Yeah, we solve it. In my projects I use standart Google WebRTC lib. And it use VideoCapturer class to provide frames from abstract camera and WebRTC. I wrote [this](https://gist.github.com/konovalov-k/a8e834465fc26e7fd67c96d6b3af1513) to provide connection between USBCamera and WebRTC. Hope it helps.

pumatus, 2020-06-23:
@konovalov-k Thank you very much~ 大佬～

heaveninfosoft, 2020-07-15:
@acrolink  can you send me code??

acrolink, 2020-07-15:
@heaveninfosoft 

Hope this helps:

```
    void addFrame(byte[] frame) {
        try {
            final long captureTimeNs =
                    TimeUnit.MILLISECONDS.toNanos(SystemClock.elapsedRealtime());
            VideoDimensions dimensions = new VideoDimensions(camera_WIDTH, camera_HEIGHT);
            VideoFrame videoFrame = new VideoFrame(frame,
                    dimensions, VideoFrame.RotationAngle.ROTATION_0, captureTimeNs);
            videoCapturerListener.onFrameCaptured(videoFrame);
        } catch (Exception ignored) {
        }
    }
```

heaveninfosoft, 2020-07-16:
@acrolink please send me full code 
VideoDimensions ,VideoFrame  ,videoCapturerListener  what is all this???? where is main lib?

heaveninfosoft, 2020-07-22:
@pumatus  hi send me full code

pumatus, 2020-07-22:
I didn't figure it out. I gave up... - -! @heaveninfosoft