jiangdongguo/AndroidUSBCamera #223: Empty .mp4 files writed

gesdotstar, 2020-04-22:
App works fine but after "stop record" (video or picture)  only empty file are writed (size 0 ) 
Any ideas about fixing this issue ?

jiangdongguo, 2020-04-22:
which version?

gesdotstar, 2020-04-22:
ver. 2.3.4 but also ver 2.3.2 i tried some weeks ago same problem
Notice i had to put:
...
               exclude 'lib/armeabi-v7a/libuvc.so'
               exclude 'lib/armeabi-v7a/libusb100.so'
               exclude 'lib/armeabi-v7a/libUVCCamera.so'
               exclude 'lib/armeabi-v7a/libUtils.so'
               exclude 'lib/armeabi-v7a/libTxtOverlay.so'
               exclude 'lib/armeabi-v7a/libjpeg-turbo1500.so'
....
in build.gradle due to "More than one file was found with OS independent path 'lib/armeabi-v7a/libUVCCamera.so'"   buildings errors...

gesdotstar, 2020-04-22:
> 
> 
> which version?

ver. 2.3.4 but also ver 2.3.2 i tried some weeks ago same problem
Notice i had to put:
...
exclude 'lib/armeabi-v7a/libuvc.so'
exclude 'lib/armeabi-v7a/libusb100.so'
exclude 'lib/armeabi-v7a/libUVCCamera.so'
exclude 'lib/armeabi-v7a/libUtils.so'
exclude 'lib/armeabi-v7a/libTxtOverlay.so'
exclude 'lib/armeabi-v7a/libjpeg-turbo1500.so'
....
in build.gradle due to "More than one file was found with OS independent path 'lib/armeabi-v7a/libUVCCamera.so'" buildings errors...

Rainbow87, 2020-05-25:
Same problems for me. After stop recording, the .mp4 file is empty size 0.0MB

vp-007, 2020-10-29:
facing the same issue, after stop video recording it shows 0kb of size.