jiangdongguo/AndroidUSBCamera #224: com.serenegiant:common:4.1.1. 无法下载，编译过不去。。。

jjjjbababy, 2020-04-23:
ERROR: Unable to resolve dependency for ':libusbcamera@debug/compileClasspath': Could not resolve com.serenegiant:common:4.1.1.

kevin-lemon, 2020-04-23:
同样，我也碰到了，我打算把库下载下来，再把这个包下载下来手动配置了。

jjjjbababy, 2020-04-23:
> 同样，我也碰到了，我打算把库下载下来，再把这个包下载下来手动配置了。

我也试了把 aar直接放在libs下，也不管用。。

kevin-lemon, 2020-04-23:
> > 同样，我也碰到了，我打算把库下载下来，再把这个包下载下来手动配置了。
> 
> 我也试了把 aar直接放在libs下，也不管用。。

我估计把他的这个项目的库下载到本地，然后修改他引用包里的配置，配置成本地AAR就好了

kevin-lemon, 2020-04-23:
> > > 同样，我也碰到了，我打算把库下载下来，再把这个包下载下来手动配置了。
> > 
> > 
> > 我也试了把 aar直接放在libs下，也不管用。。
> 
> 我估计把他的这个项目的库下载到本地，然后修改他引用包里的配置，配置成本地AAR就好了

搞定

jjjjbababy, 2020-04-23:
> > > > 同样，我也碰到了，我打算把库下载下来，再把这个包下载下来手动配置了。
> > > 
> > > 
> > > 我也试了把 aar直接放在libs下，也不管用。。
> > 
> > 
> > 我估计把他的这个项目的库下载到本地，然后修改他引用包里的配置，配置成本地AAR就好了
> 
> 搞定

咋搞的

kevin-lemon, 2020-04-23:
> > > > > 同样，我也碰到了，我打算把库下载下来，再把这个包下载下来手动配置了。
> > > > 
> > > > 
> > > > 我也试了把 aar直接放在libs下，也不管用。。
> > > 
> > > 
> > > 我估计把他的这个项目的库下载到本地，然后修改他引用包里的配置，配置成本地AAR就好了
> > 
> > 
> > 搞定
> 
> 咋搞的

clone下来项目，吧libusbcamera 单独引用进去，然后修改libusbcamera的gradle，弄成本地aar

jjjjbababy, 2020-04-23:
> > > > > > 同样，我也碰到了，我打算把库下载下来，再把这个包下载下来手动配置了。
> > > > > 
> > > > > 
> > > > > 我也试了把 aar直接放在libs下，也不管用。。
> > > > 
> > > > 
> > > > 我估计把他的这个项目的库下载到本地，然后修改他引用包里的配置，配置成本地AAR就好了
> > > 
> > > 
> > > 搞定
> > 
> > 
> > 咋搞的
> 
> clone下来项目，吧libusbcamera 单独引用进去，然后修改libusbcamera的gradle，弄成本地aar

我把那个common-2.12.4.aar  下载下来了 放到libs里的 ，
implementation fileTree(dir: 'libs', include: ['*.jar','*.aar'])这个写了 ，还是不管用

jjjjbababy, 2020-04-23:
implementation fileTree(dir: 'libs', include: ['*.jar','*.aar'])

kevin-lemon, 2020-04-23:
> > > > > > > 同样，我也碰到了，我打算把库下载下来，再把这个包下载下来手动配置了。
> > > > > > 
> > > > > > 
> > > > > > 我也试了把 aar直接放在libs下，也不管用。。
> > > > > 
> > > > > 
> > > > > 我估计把他的这个项目的库下载到本地，然后修改他引用包里的配置，配置成本地AAR就好了
> > > > 
> > > > 
> > > > 搞定
> > > 
> > > 
> > > 咋搞的
> > 
> > 
> > clone下来项目，吧libusbcamera 单独引用进去，然后修改libusbcamera的gradle，弄成本地aar
> 
> 我把那个common-2.12.4.aar 下载下来了 放到libs里的 ，
> implementation fileTree(dir: 'libs', include: ['_.jar','_.aar'])这个写了 ，还是不管用

你是放到libusbcamera下面？

jjjjbababy, 2020-04-23:
> 你是放到libusbcamera下面？

是啊，libusbcamera下  src下 的libs里

kevin-lemon, 2020-04-23:
> 是啊，libusbcamera下 src下 的libs里

我刚编译通过的。加我QQ765002108

jjjjbababy, 2020-05-05:
solved

closed 2020-05-05 by jjjjbababy

Dreamskya, 2021-03-22:
@jjjjbababy 怎么搞定的，我也是你这种做法，还是不行。