jiangdongguo/AndroidUSBCamera #233: setOnPreviewFrameListener not working

faizuuuuu, 2020-05-29:
mCameraHelper.setOnPreviewFrameListener
this method is not getting called .Not able to get the fram of the video.

jiangdongguo, 2020-06-11:
how about change you camera?

faizuuuuu, 2020-06-11:
i changed camera as well .
this method was working before .
Now frames listener is not getting called on connecting camera .

jiangdongguo, 2020-06-15:
I see. Let me check it.

heaveninfosoft, 2020-06-28:
> I see. Let me check it.

how to do dual USB Camera??

JCCAbrunhosa, 2020-08-21:
Any news regarding this issue? I can't seem to get the listener to work. I want to capture the frame but nothing happens.

kanstantsin, 2020-09-11:
Got same issue. For me the problem was in following piece of [code](https://github.com/jiangdongguo/AndroidUSBCamera/blob/master/libusbcamera/src/main/java/com/serenegiant/usb/common/AbstractUVCCameraHandler.java#L547).
```
            try {
                mUVCCamera.setPreviewSize(mWidth, mHeight, 1, 31, mPreviewMode, mBandwidthFactor);
                // 获取USB Camera预览数据，使用NV21颜色会失真
                // 无论使用YUV还是MPEG，setFrameCallback的设置效果一致
//				mUVCCamera.setFrameCallback(mIFrameCallback, UVCCamera.PIXEL_FORMAT_NV21);
                mUVCCamera.setFrameCallback(mIFrameCallback, UVCCamera.PIXEL_FORMAT_YUV420SP);
            } catch (final IllegalArgumentException e) {
                try {
                    // fallback to YUV mode
                    mUVCCamera.setPreviewSize(mWidth, mHeight, 1, 31, UVCCamera.DEFAULT_PREVIEW_MODE, mBandwidthFactor);
                } catch (final IllegalArgumentException e1) {
                    callOnError(e1);
                    return;
                }
            }
```

I got exception at
`mUVCCamera.setPreviewSize(mWidth, mHeight, 1, 31, mPreviewMode, mBandwidthFactor);`
and jumped to catch block:
`mUVCCamera.setPreviewSize(mWidth, mHeight, 1, 31, UVCCamera.DEFAULT_PREVIEW_MODE, mBandwidthFactor);`
which uses `UVCCamera.DEFAULT_PREVIEW_MODE ` but does not set frame callback.
Inserting `mUVCCamera.setFrameCallback(mIFrameCallback, UVCCamera.PIXEL_FORMAT_YUV420SP);` after this line fixed callback issue for me.

@jiangdongguo, looks like a bug. If so, is my fix correct?

wcj85422107, 2022-03-01:
> Got same issue. For me the problem was in following piece of [...]

Have you solved it