jiangdongguo/AndroidUSBCamera #234: JNI libs source

yshteinm, 2020-05-30:
Hello, where is JNI libs source? Is it exactly the same as saki4510 or you made any modifications in cpp files?

jiangdongguo, 2020-06-11:
please provider you email.

heaveninfosoft, 2020-07-15:
Hello, where is JNI libs source? Is it exactly the same as saki4510 or you made any modifications in cpp files?  i want to know how to create 
can you take video and pic at same time for two camera??
i developing app that should show two camera same time
also with image capture,video record and video streaming

heaveninfosoft, 2020-07-15:
@jiangdongguo 
can you take video and pic at same time for two camera??
i developing app that should show two camera same time
also with image capture,video record and video streaming

can you suggest me any demo with this all code for dual usb camera?

2ndtonone, 2020-08-12:
> please provider you email.

1002509020@qq.com
遇到stopPreview 系统崩溃，希望大神提供下你的jni模块代码。万分感谢

raphaelm, 2021-01-14:
Did you ever receive the sources?

yshteinm, 2021-01-14:
No, I used sources from saki4510