jiangdongguo/AndroidUSBCamera #242: surface should be one of SurfaceHolder, Surface or SurfaceTexture: null

tsubasa-kun, 2020-06-23:
布局和类文件都是按照你的源码照搬写出来的，可是每次运行都会报错崩溃，就是报“java.lang.IllegalArgumentException: surface should be one of SurfaceHolder, Surface or SurfaceTexture: null”，我一直没找到问题所在，请问知道这是什么原因吗？

Paircheng, 2020-07-13:
请问这个问题解决了吗？我也遇到了相同的问题

tsubasa-kun, 2020-07-13:
> 请问这个问题解决了吗？我也遇到了相同的问题

这个原版的我没找到解决办法，我从别的项目里扒了一个用的

Paircheng, 2020-07-18:
> > 请问这个问题解决了吗？我也遇到了相同的问题
> 
> 这个原版的我没找到解决办法，我从别的项目里扒了一个用的

请问一下是如何解决的呢？

tsubasa-kun, 2020-07-24:
> > > 请问这个问题解决了吗？我也遇到了相同的问题
> > 
> > 
> > 这个原版的我没找到解决办法，我从别的项目里扒了一个用的
> 
> 请问一下是如何解决的呢？

从别人的可运行的项目，把引用的库复制下来。没有用这个

quocnhat7, 2020-08-14:
You can download the source code as ZIP, create a module and see the source in Android studio.
Then, search this "surface should be one of SurfaceHolder" in the source, you will know the reason.

Pls make sure your interface video-display is an expected view, you can not start preview in another view.

Huan2015, 2021-01-14:
有解决的吗？demo运行没问题，但是我把USBCameraActivity都复制到自己的项目就不行了，build文件啥的都照着demo来的，还是不行

Huan2015, 2021-01-14:
找到问题了，Manifest中application下边配置了android:hardwareAccelerated="false"  android:largeHeap="true"

KanwalKhan-20, 2021-04-28:
Any solution to this problem?

aieyes, 2022-02-25:
> 找到问题了，Manifest中application下边配置了android:hardwareAccelerated="false" android:largeHeap="true"

应该是demo写得有问题，我这边是重启设备有时才会抛这个异常然后奔溃一次，initUSBMonitor 里面自动执行了预览，但是Surface此时又为null。没等它创建完就调用预览了。我修改了执行顺序就没问题了：
mCameraHelper = UVCCameraHelper.getInstance();
        mCameraHelper.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_MJPEG);
        if (mCameraHelper.getUSBMonitor() == null) {
            mUVCCameraView = view.findViewById(R.id.cameraView);
            mUVCCameraView.setCallback(new CameraViewInterface.Callback() {
                @Override
                public void onSurfaceCreated(CameraViewInterface view, Surface surface) {
                    Log.d(TAG, "camera surface created");
                    mCameraHelper.initUSBMonitor(requireActivity(), view, cameraListener);// 在这里初始化usbMonitor
                    mCameraHelper.registerUSB();
                }

                @Override
                public void onSurfaceChanged(CameraViewInterface view, Surface surface, int width, int height) {
                    Log.d(TAG, "camera surface changed");
                }

                @Override
                public void onSurfaceDestroy(CameraViewInterface view, Surface surface) {
                    Log.d(TAG, "camera surface destroy");
                }
            });
        }