jiangdongguo/AndroidUSBCamera #245: 使用分辨率480*640，预览出现卡顿问题，怎么解决

Paircheng, 2020-07-04:
使用分辨率480*640，预览出现卡顿问题，怎么解决？提高了摄像头的清晰度

Paircheng, 2020-07-04:
抖动的时候预览切屏问题，android 7.0

Paircheng, 2020-07-07:
停止预览是出现APP停止运行，报错如下：
07-06 15:53:47.212 I/AEE_AED ( 5378): signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
07-06 15:53:47.221 I/AEE_AED ( 5378): Abort message: 'art/runtime/java_vm_ext.cc:496] JNI DETECTED ERROR IN APPLICATION: jmethodID was NULL'
07-06 15:53:47.222 I/AEE_AED ( 5378):     r0 00000000  r1 000014fc  r2 00000006  r3 00000008
07-06 15:53:47.222 I/AEE_AED ( 5378):     r4 88b76978  r5 00000006  r6 88b76920  r7 0000010c
07-06 15:53:47.222 I/AEE_AED ( 5378):     r8 00000000  r9 ac6689f4  sl 000001c2  fp ac62e13f
07-06 15:53:47.222 I/AEE_AED ( 5378):     ip 00000000  sp 88b76440  lr ae5534e7  pc ae555d44  cpsr 200f0010
07-06 15:53:47.279 I/AEE_AED ( 5378): 
07-06 15:53:47.279 I/AEE_AED ( 5378): backtrace:
07-06 15:53:47.279 I/AEE_AED ( 5378):     #00 pc 0004ad44  /system/lib/libc.so (tgkill+12)
07-06 15:53:47.279 I/AEE_AED ( 5378):     #01 pc 000484e3  /system/lib/libc.so (pthread_kill+34)
07-06 15:53:47.279 I/AEE_AED ( 5378):     #02 pc 0001d495  /system/lib/libc.so (raise+10)
07-06 15:53:47.280 I/AEE_AED ( 5378):     #03 pc 00018fd1  /system/lib/libc.so (__libc_android_abort+34)
07-06 15:53:47.280 I/AEE_AED ( 5378):     #04 pc 00017034  /system/lib/libc.so (abort+4)
07-06 15:53:47.280 I/AEE_AED ( 5378):     #05 pc 00319825  /system/lib/libart.so (_ZN3art7Runtime5AbortEv+252)
07-06 15:53:47.280 I/AEE_AED ( 5378):     #06 pc 000b4d39  /system/lib/libart.so (_ZN3art10LogMessageD2Ev+864)
07-06 15:53:47.280 I/AEE_AED ( 5378):     #07 pc 00238991  /system/lib/libart.so (_ZN3art9JavaVMExt8JniAbortEPKcS2_+2572)
07-06 15:53:47.280 I/AEE_AED ( 5378):     #08 pc 0023940f  /system/lib/libart.so (_ZN3art9JavaVMExt9JniAbortVEPKcS2_St9__va_list+58)
07-06 15:53:47.280 I/AEE_AED ( 5378):     #09 pc 000ca6db  /system/lib/libart.so (_ZN3art11ScopedCheck6AbortFEPKcz+46)
07-06 15:53:47.281 I/AEE_AED ( 5378):     #10 pc 000ca123  /system/lib/libart.so (_ZN3art11ScopedCheck13CheckMethodIDERNS_18ScopedObjectAccessEP10_jmethodID+118)
07-06 15:53:47.281 I/AEE_AED ( 5378):     #11 pc 000c92ff  /system/lib/libart.so (_ZN3art11ScopedCheck22CheckPossibleHeapValueERNS_18ScopedObjectAccessEcNS_12JniValueTypeE+58)
07-06 15:53:47.281 I/AEE_AED ( 5378):     #12 pc 000c87bb  /system/lib/libart.so (_ZN3art11ScopedCheck5CheckERNS_18ScopedObjectAccessEbPKcPNS_12JniValueTypeE+802)
07-06 15:53:47.281 I/AEE_AED ( 5378):     #13 pc 000cc967  /system/lib/libart.so (_ZN3art8CheckJNI13CheckCallArgsERNS_18ScopedObjectAccessERNS_11ScopedCheckEP7_JNIEnvP8_jobjectP7_jclassP10_jmethodIDNS_10InvokeTypeEPKNS_7VarArgsE+114)
07-06 15:53:47.281 I/AEE_AED ( 5378):     #14 pc 000cbf95  /system/lib/libart.so (_ZN3art8CheckJNI11CallMethodVEPKcP7_JNIEnvP8_jobjectP7_jclassP10_jmethodIDSt9__va_listNS_9Primitive4TypeENS_10InvokeTypeE+512)
07-06 15:53:47.281 I/AEE_AED ( 5378):     #15 pc 000c2169  /system/lib/libart.so (_ZN3art8CheckJNI15CallVoidMethodVEP7_JNIEnvP8_jobjectP10_jmethodIDSt9__va_list+44)
07-06 15:53:47.282 I/AEE_AED ( 5378):     #16 pc 000114f0  /data/app/com.veclink.vecsipsimple-1/lib/arm/libUVCCamera.so (_ZN7_JNIEnv14CallVoidMethodEP8_jobjectP10_jmethodIDz+52)
07-06 15:53:47.282 I/AEE_AED ( 5378):     #17 pc 0001140c  /data/app/com.veclink.vecsipsimple-1/lib/arm/libUVCCamera.so (_ZN10UVCPreview19do_capture_callbackEP7_JNIEnvP9uvc_frame+244)
07-06 15:53:47.282 I/AEE_AED ( 5378):     #18 pc 000110bc  /data/app/com.veclink.vecsipsimple-1/lib/arm/libUVCCamera.so (_ZN10UVCPreview10do_captureEP7_JNIEnv+144)
07-06 15:53:47.282 I/AEE_AED ( 5378):     #19 pc 00010b04  /data/app/com.veclink.vecsipsimple-1/lib/arm/libUVCCamera.so (_ZN10UVCPreview19capture_thread_funcEPv+64)
07-06 15:53:47.282 I/AEE_AED ( 5378):     #20 pc 00047fb3  /system/lib/libc.so (_ZL15__pthread_startPv+22)
07-06 15:53:47.282 I/AEE_AED ( 5378):     #21 pc 00019a1d  /system/lib/libc.so (__start_thread+6)

so文件报错，请看一下怎么修改，谢谢！

quocnhat7, 2020-08-14:
You should get supported_preview_size of your external camera.
Then check whether your target size is in that list or not. We should not use a fixed size which camera does not support.
If that size is not supported, you can find nearest size in that list to apply.