jiangdongguo/AndroidUSBCamera #251: Error "jetified-common-4.1.1\res\values\values.xml:687:5-705:22: AAPT: error: duplicate value for resource 'attr/orientation' with config ''."

zipswich, 2020-07-20:
As soon as I add the following to  build.gradle:
`implementation 'com.github.jiangdongguo:AndroidUSBCamera:2.3.4'`
building an app generates the following error:
`jetified-common-4.1.1\res\values\values.xml:687:5-705:22: AAPT: error: duplicate value for resource 'attr/orientation' with config ''.`

Yes, I have taken a look at [this SO question](https://stackoverflow.com/questions/53895569/error-duplicate-value-for-resource-attr-orientation-with-config-aapt-v2-a), but the solution does not work.

The app works perfectly without using jiangdongguo:AndroidUSBCamera.   I am not sure this is necessarily the library's issue, but I hope to get a tip here to address it.

mobileDev555, 2020-11-08:
I think it is related to a resource error mainly.
First, Please do "Invalidate caches / Restart" in your android studio and then if it does not fix your problem then please downgrade the App/Gradle libraries.

IvanAbacumov, 2020-11-08:
@zipswich
 I have the latest version of Gradle installed, I added a library, added NDK and everything works without errors

shiduan, 2020-12-07:
I also encounter problems that "duplicate value for resource 'attr/orientation' with config '' and how did you deal with it?

mobileDev555, 2020-12-07:
Thank you for your email.
If you need my help, Please contact me on Skype.

my skype : live:.cid.e5675d33daad7e3f

kevengh, 2021-03-07:
who had deal whit "duplicate value for resource 'attr/orientation' with config "??

ricrebelo, 2021-05-28:
I have the same issue, any fix ?