jiangdongguo/AndroidUSBCamera #258: 可以支持USB的音频输入吗？

butterfly0Dream, 2020-08-13:


quocnhat7, 2020-08-14:
Good question.  but perhaps we need to ask it in parent project
[https://github.com/saki4510t/UVCCamera](url)

I am also finding more info about this topic.

quocnhat7, 2020-08-26:
That is what I want to know before. But now my project does not need the audio. So I dont know about solution for your question now.

jiangdongguo, 2020-09-11:
支持

CarryHekangkai, 2020-09-11:
同样想知道如何播放摄像头采集的音频

butterfly0Dream, 2020-09-12:
> 支持

请问大佬什么时候更新usb音频部分呢？另外视频格式可以增加h264的支持吗？

sduqlsc, 2020-11-13:
@jiangdongguo 求指点，我也遇到了这个问题，摄像头带有麦克风，推流之后没有声音，这个需要改造库里的哪些地方？

sduqlsc, 2020-11-16:
大抵是要自己写 NDK 的驱动

CarryHekangkai, 2020-11-16:
> 大抵是要自己写 NDK 的驱动

改一下源码就可以了，原本功能有把USB的音频保存成文件

sduqlsc, 2020-11-17:
> > 大抵是要自己写 NDK 的驱动
> 
> 改一下源码就可以了，原本功能有把USB的音频保存成文件

您说的是要修改 libusb 吗还是本库的？

wujianwei121, 2020-12-08:
> 大抵是要自己写 NDK 的驱动

朋友  你推流没有声音的问题解决了嘛  我目前也遇到这个问题

wujianwei121, 2020-12-09:
> > 大抵是要自己写 NDK 的驱动
> 
> 改一下源码就可以了，原本功能有把USB的音频保存成文件

兄弟  你是怎么改的  能不能告诉我一下

CarryHekangkai, 2020-12-09:
> > > 大抵是要自己写 NDK 的驱动
> > 
> > 
> > 改一下源码就可以了，原本功能有把USB的音频保存成文件
> 
> 兄弟 你是怎么改的 能不能告诉我一下

把UVCCamera的源码下载下来，在MediaAudioEncoder类的AudioThread处理获取到的音频数据。
`if (audioRecord != null) {
				try {
					if (mIsCapturing) {
						if (DEBUG) Log.v(TAG, "AudioThread:start audio recording");
						int readBytes;
						audioRecord.startRecording();
						try {
							for ( ; mIsCapturing && !mRequestStop && !mIsEOS ; ) {
								// read audio data from internal mic
								buf.clear();
								try {
//									readBytes = audioRecord.read(buf, SAMPLES_PER_FRAME);

									readBytes = audioRecord.read(buffer, 0, SAMPLES_PER_FRAME);
									outputStream.write(buffer);
//									outstream.write(buffer, 0, SAMPLES_PER_FRAME);
//									buf.get(buffer, 0, SAMPLES_PER_FRAME);

								} catch (final Exception e) {
									break;
								}
								if (readBytes > 0) {
									// set audio data to encoder
//									buf.position(readBytes);
//									buf.flip();

//									encode(buf, readBytes, getPTSUs());
									frameAvailableSoon();
									cnt++;
								}
							}
							if (cnt > 0) {
								frameAvailableSoon();
							}
						} finally {
							outputStream.close();
							audioRecord.stop();
//							outstream.close();
						}
					}
				} catch (final Exception e) {
					Log.e(TAG, "AudioThread#run", e);
				} finally {
					audioRecord.release();
				}
			}
`
`
private final MediaEncoder.MediaEncoderListener mMediaEncoderListener = new MediaEncoder.MediaEncoderListener() {
			@Override
			public void onPrepared(final MediaEncoder encoder) {
				if (DEBUG) Log.v(TAG, "onPrepared:encoder=" + encoder);
				mIsRecording = true;
				if (encoder instanceof MediaVideoEncoder) {
					try {
						mWeakCameraView.get().setVideoEncoder((MediaVideoEncoder) encoder);
					} catch (final Exception e) {
						Log.e(TAG, "onPrepared:", e);
					}
				}
				if (encoder instanceof MediaSurfaceEncoder){
					try {
						mWeakCameraView.get().setVideoEncoder((MediaSurfaceEncoder)encoder);
						mUVCCamera.startCapture(((MediaSurfaceEncoder)encoder).getInputSurface());
					} catch (final Exception e) {
						Log.e(TAG, "onPrepared:", e);
					}
				}
				if (encoder instanceof MediaAudioEncoder){
					path = ((MediaAudioEncoder) encoder).getFilePath();
					startRecord();
				}
			}
`

wujianwei121, 2020-12-09:
> mMediaEncoderListener

感谢兄弟   能不能把源码打包发一份给我看看  417757407@qq.com

wujianwei121, 2020-12-09:
> 把UVCCamera的源码下载下来，在MediaAudioEncoder类的AudioThread处理获取到的音频数据。
> `if (audioRecord != null) {
> try {
> if (mIsCapturing) {
> if (DEBUG) Log.v(TAG, "AudioThread:start audio recording");
> int readBytes; [...]

感谢兄弟 能不能把源码打包发一份给我看看 417757407@qq.com

wujianwei121, 2020-12-09:
兄弟  能不能把这个文件整个发给我   谢谢了   困扰我小半个月了  

    

  
  
  
 for ( ; mIsCapturing &amp;&amp; !mRequestStop &amp;&amp; !mIsEOS ; ) {
 							} catch (final Exception e) { 								break; 							} 							if (readBytes &gt; 0) { 								// set audio data to encoder  
 
 if (cnt &gt; 0) {

butterfly0Dream, 2020-12-09:
> > > > > 大抵是要自己写 NDK 的驱动
> > > > 
> > > > 
> > > > 改一下源码就可以了，原本功能有把USB的音频保存成文件
> > > 
> > > 
> > > 兄弟 你是怎么改的 能不能告诉我一下
> > 
> > 
> > 把UVCCamera的源码下载下来，在MediaAudioEncoder类的AudioThread处理获取到的音频数据。
> > `if (audioRecord != null) {
> > try {
> > if (mIsCapturing) {
> > if (DEBUG) Log.v(TAG, "AudioThread:start audio recording");
> > int readBytes;
> > audioRecord.startRecording();
> > try {
> > for ( ; mIsCapturing && !mRequestStop && !mIsEOS ; ) {
> > // read audio data from internal mic
> > buf.clear();
> > try {
> > // readBytes = audioRecord.read(buf, SAMPLES_PER_FRAME);
> > ```
> > 								readBytes = audioRecord.read(buffer, 0, SAMPLES_PER_FRAME);
> > 								outputStream.write(buffer);
> > ```
> > 
> > 
> > // outstream.write(buffer, 0, SAMPLES_PER_FRAME);
> > // buf.get(buffer, 0, SAMPLES_PER_FRAME);
> > ```
> > 							} catch (final Exception e) {
> > 								break;
> > 							}
> > 							if (readBytes > 0) {
> > 								// set audio data to encoder
> > ```
> > 
> > 
> > // buf.position(readBytes);
> > // buf.flip();
> > // encode(buf, readBytes, getPTSUs());
> > frameAvailableSoon();
> > cnt++;
> > }
> > }
> > if (cnt > 0) {
> > frameAvailableSoon();
> > }
> > } finally {
> > outputStream.close();
> > audioRecord.stop();
> > // outstream.close();
> > }
> > }
> > } catch (final Exception e) {
> > Log.e(TAG, "AudioThread#run", e);
> > } finally {
> > audioRecord.release();
> > }
> > }
> > ` `
> > private final MediaEncoder.MediaEncoderListener mMediaEncoderListener = new MediaEncoder.MediaEncoderListener() {
> > @override
> > public void onPrepared(final MediaEncoder encoder) {
> > if (DEBUG) Log.v(TAG, "onPrepared:encoder=" + encoder);
> > mIsRecording = true;
> > if (encoder instanceof MediaVideoEncoder) {
> > try {
> > mWeakCameraView.get().setVideoEncoder((MediaVideoEncoder) encoder);
> > } catch (final Exception e) {
> > Log.e(TAG, "onPrepared:", e);
> > }
> > }
> > if (encoder instanceof MediaSurfaceEncoder){
> > try {
> > mWeakCameraView.get().setVideoEncoder((MediaSurfaceEncoder)encoder);
> > mUVCCamera.startCapture(((MediaSurfaceEncoder)encoder).getInputSurface());
> > } catch (final Exception e) {
> > Log.e(TAG, "onPrepared:", e);
> > }
> > }
> > if (encoder instanceof MediaAudioEncoder){
> > path = ((MediaAudioEncoder) encoder).getFilePath();
> > startRecord();
> > }
> > }
> > `
> 
> 感谢兄弟 能不能把源码打包发一份给我看看 [417757407@qq.com](mailto:417757407@qq.com)

这个AudioThread是使用AudioRecord获取手机系统录音。只不过凑巧是，audioSource设置为mic，且usb摄像头有mic，那么拿到的录音数据就是摄像头mic的（类似外置录音源）。但是这样有一个问题就是，app会一直占用系统录音，其他app就无法调用系统录音了。要想直接拿到摄像头的音频数据，必须从NDK层入手。

wujianwei121, 2020-12-09:
了解  了解   写了  大兄弟

Flame109, 2021-05-14:
> 了解  了解   写了  大兄弟
> […](#)

大兄弟 这个问题你后来解决了吗

MAYBreath, 2021-06-07:
https://app.mi.com/details?id=com.shenyaocn.android.usbcamera
这个支持音频输入 不知道他怎么做到的

MAYBreath, 2021-06-08:
> 支持

不支持吧，是说和预览视频一样能同时播放声音，不是录制声音，比如采集卡直接接入电视盒子，直接通过设备看电影能放声音之类的。

sduqlsc, 2021-06-18:
> https://app.mi.com/details?id=com.shenyaocn.android.usbcamera
> 这个支持音频输入 不知道他怎么做到的

微博可以找到他

MAYBreath, 2021-06-22:
> > https://app.mi.com/details?id=com.shenyaocn.android.usbcamera
> > 这个支持音频输入 不知道他怎么做到的
> 
> 微博可以找到他

谢谢，是我肤浅了 加个trackaudio直接播放就可以了。

AndroidBody, 2022-05-20:
有解决了吗？各位

EthanCo, 2022-07-21:
我已解决，可以看我的博客 : https://blog.csdn.net/EthanCo/article/details/125802505

MAYBreath, 2022-10-09:
> 我已解决，可以看我的博客 : https://blog.csdn.net/EthanCo/article/details/125802505

其实不需要额外用库的，他本身就自带有AAC编码器AACEncodeConsumer 里面就有采集音频 直接加载就能播放

不过真巧 我也是打算用在车机上，不知道你是哪个车企的？

keep2iron, 2023-11-04:
已解决，

keep2iron, 2023-11-04:
https://github.com/jim0608/android_usbaudio

keep2iron, 2023-11-04:
https://github.com/jim0608/android_usbaudio 
https://github.com/saki4510t/UVCCamera
参考以上项目将，audio移植到uvc camera，可解决预览输出音频问题

HeLiHuang, 2024-06-04:
> https://github.com/jim0608/android_usbaudio https://github.com/saki4510t/UVCCamera 参考以上项目将，audio移植到uvc camera，可解决预览输出音频问题

两者结合在一起，usb设备拔出就闪退

cristiano-linvix, 2024-07-09:
> 这个AudioThread是使用AudioRecord获取手机系统录音。只不过凑巧是，audioSource设置为mic，且usb摄像头有mic，那么拿到的录音数据就是摄像头mic的（类似外置录音源）。但是这样有一个问题就是，app会一直占用系统录音，其他app就无法调用系统录音了。要想直接拿到摄像头的音频数据，必须从NDK层入手。

Can you get a code example of how to do this?