jiangdongguo/AndroidUSBCamera #264: How to get around "setDefaultFrameFormat should be call before initMonitor" + Info SDK 28

OlavAa, 2020-08-25:
First of all I like to inform you that I just successfully built an app with target SDK = 28. I had given up and started to look into the Camera2 API when I suddenly realized that the web cam video was indeed working.
The change: I added the following permission to the manifest:
`<uses-permission android:name="android.permission.CAMERA" />`

My problem is this:
My app has a list of widgets that include small features that can be expanded. Different widgets for mjpg video streaming from an URL and for USB. When user decides to he may click the small video preview and expand the video into a separate activity. The problem I cannot get around is "public void setDefaultFrameFormat" in UVCCameraHelper that throws an exception if it is executed before the initMonitor. The widget and the separate video activity basically runs the code in the same fashion as described in your example app.
In other words: The "mCameraHelper.initUSBMonitor" run for the initial small video preview somehow is still active when my second activity tries to run the same code. I even put in a delay so that the initial fragment reached onApiDisconnected. But no way to avoid the error for the new activity starting up.
I finally got around it by unloading the widget completely before I started the new activity. But that obviously opens a new challenge since I want the user to go back (away from) the big video activity and still have the old list of widgets available.

Do you have any suggestions on how to avoid this situation before I start the second activity?

closed 2020-09-12 by OlavAa

j0hnw0rk3r, 2022-01-30:
@OlavAa Hi, have you solved this issue? I have the same problem here. Can you please share your workaround? Thank you!

OlavAa, 2022-01-30:
I could not get around it directly. Instead I closed the video widget and started the bigger video window with a delay.
I have abandoned this solution later and only use a single video window.
App now build using SDK30. My current challenge is that the user needs to:

1. Plug the USB cable in BEFORE starting my app. Using a device filter, my app is presented as a choice to let the user start the app. This way, everything works
2. The user may rip out the plug whike the app is running without causing a crash
3. But the USB cable cannot be plugged in afterwards, the app crashes and so far I have not found away around it. The crash is caused by mission permission (which is granted when starting the app AFTER putting the cable in, but somehow is not detected to be able to request if cable is plugged in afterwards)