jiangdongguo/AndroidUSBCamera #270: native crash

gbk-t, 2020-09-15:
2020-09-14 17:00:13.783  4978  8360 V CameraThread: handleStartPreview:
2020-09-14 17:00:13.784  4978  8360 W libuvc/stream: [8360*stream.c:270:uvc_query_stream_ctrl]:fix up block for cameras that fail to set dwMax
2020-09-14 17:00:13.785  4978  8360 W libuvc/stream: [8360*stream.c:270:uvc_query_stream_ctrl]:fix up block for cameras that fail to set dwMax
2020-09-14 17:00:13.786  4978  8360 W libuvc/stream: [8360*stream.c:270:uvc_query_stream_ctrl]:fix up block for cameras that fail to set dwMax
2020-09-14 17:00:13.786  4978  8360 W libuvc/stream: [8360*stream.c:270:uvc_query_stream_ctrl]:fix up block for cameras that fail to set dwMax
2020-09-14 17:00:13.787  4978  8360 W libuvc/stream: [8360*stream.c:270:uvc_query_stream_ctrl]:fix up block for cameras that fail to set dwMax
2020-09-14 17:00:13.787  4978  8360 W libuvc/stream: [8360*stream.c:270:uvc_query_stream_ctrl]:fix up block for cameras that fail to set dwMax
2020-09-14 17:00:13.792  4978  8360 I libUVCCamera: [8360*UVCPreview.cpp:276:callbackPixelFormatChanged]:PIXEL_FORMAT_YUV20SP:
2020-09-14 17:00:13.804  4978  8372 I libUVCCamera: [8372*UVCPreview.cpp:492:prepare_preview]:frameSize=(1920,1080)@MJPEG
--------- beginning of crash
2020-09-14 17:00:13.805  4978  8370 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x0 in tid 8370 (CameraThread)
2020-09-14 17:00:13.812   249   249 W         : debuggerd: handling request: pid=4978 uid=1000 gid=1000 tid=8370
2020-09-14 17:00:13.914  8373  8373 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2020-09-14 17:00:13.915  8373  8373 F DEBUG   : Build fingerprint: 'qcom/ST2303/ST2303:7.1.2/N2G47H/senter09071625:userdebug/'
2020-09-14 17:00:13.915  8373  8373 F DEBUG   : Revision: '0'
2020-09-14 17:00:13.915  8373  8373 F DEBUG   : ABI: 'arm'
2020-09-14 17:00:13.916  8373  8373 F DEBUG   : pid: 4978, tid: 8370, name: CameraThread  >>> com.senter.stmonitor <<<
2020-09-14 17:00:13.916  8373  8373 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
2020-09-14 17:00:13.916  8373  8373 F DEBUG   :     r0 a02d7274  r1 00000000  r2 0000a6db  r3 00000001
2020-09-14 17:00:13.916  8373  8373 F DEBUG   :     r4 a02d7250  r5 a02d7274  r6 00000000  r7 00000002
2020-09-14 17:00:13.916  8373  8373 F DEBUG   :     r8 92a2e1c0  r9 94283400  sl a02d7270  fp 905fa610
2020-09-14 17:00:13.916  8373  8373 F DEBUG   :     ip 92cbedc0  sp 905fa608  lr 92caf318  pc 92cb0bec  cpsr a00d0010
2020-09-14 17:00:13.929  8373  8373 F DEBUG   : 
2020-09-14 17:00:13.929  8373  8373 F DEBUG   : backtrace:
2020-09-14 17:00:13.929  8373  8373 F DEBUG   :     #00 pc 0000dbec  /data/app/com.senter.stmonitor-1/lib/arm/libusb100.so
2020-09-14 17:00:13.929  8373  8373 F DEBUG   :     #01 pc 0000c314  /data/app/com.senter.stmonitor-1/lib/arm/libusb100.so (usbi_handle_transfer_completion+320)
2020-09-14 17:00:13.929  8373  8373 F DEBUG   :     #02 pc 000127a0  /data/app/com.senter.stmonitor-1/lib/arm/libusb100.so
2020-09-14 17:00:13.929  8373  8373 F DEBUG   :     #03 pc 0000d200  /data/app/com.senter.stmonitor-1/lib/arm/libusb100.so
2020-09-14 17:00:13.929  8373  8373 F DEBUG   :     #04 pc 0000cb60  /data/app/com.senter.stmonitor-1/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+968)
2020-09-14 17:00:13.930  8373  8373 F DEBUG   :     #05 pc 0000d3cc  /data/app/com.senter.stmonitor-1/lib/arm/libusb100.so (libusb_handle_events+52)
2020-09-14 17:00:13.930  8373  8373 F DEBUG   :     #06 pc 00012c20  /data/app/com.senter.stmonitor-1/lib/arm/libuvc.so (_uvc_handle_events+136)
2020-09-14 17:00:13.930  8373  8373 F DEBUG   :     #07 pc 00047073  /system/lib/libc.so (_ZL15__pthread_startPv+22)
2020-09-14 17:00:13.930  8373  8373 F DEBUG   :     #08 pc 00019d7d  /system/lib/libc.so (__start_thread+6)
2020-09-14 17:00:14.630  5839  5839 D HeadsetPhoneState: Enter onSignalStrengthsChanged
2020-09-14 17:00:14.630  5839  5839 D HeadsetPhoneState: Exit onSignalStrengthsChanged

jiangdongguo, 2020-09-25:
please update to new and have a try

Paircheng, 2020-10-19:
> please update to new and have a try

I have a similar issue, the log is as follows：

2020-10-19 14:18:15.928 7608-8410/com.chauncy.demo W/libuvc/stream: [8410*stream.c:270:uvc_query_stream_ctrl]:fix up block for cameras that fail to set dwMax
2020-10-19 14:18:15.928 7608-8410/com.chauncy.demo W/libuvc/stream: [8410*stream.c:270:uvc_query_stream_ctrl]:fix up block for cameras that fail to set dwMax
2020-10-19 14:18:15.930 7608-8410/com.chauncy.demo W/libuvc/stream: [8410*stream.c:270:uvc_query_stream_ctrl]:fix up block for cameras that fail to set dwMax

I used the lastest version.

Jackie-ZJW, 2021-11-22:
> I have a similar issue, the log is as follows：
>
> 2020-10-19 14:18:15.928 7608-8410/com.chauncy.demo W/libuvc/stream: [8410_stream.c:270:uvc_query_stream_ctrl]:fix up block for cameras that fail to [...]

hi, bro, Have you solved the problem “fix up block for cameras that fail to set dwMax” ？ I still had this problem with the latest version AndroidUSBCamera。