jiangdongguo/AndroidUSBCamera #274: Unable to get preview on a fragment

Vishalratnam, 2020-09-25:
I have a fragment inside which I'm trying to show the camera preview.
I get the following error on the Logcat:
`D/UsbDeviceConnectionJNI: close
09-25 14:51:49.469 10662-10702/com.example.anecdot I/USBMonitor: name=/dev/bus/usb/003/007,desc=35,busnum=3,devnum=7,rawDesc=[B@f5d9b1b
09-25 14:51:49.469 10662-10702/com.example.anecdot V/AbsUVCCameraHandler: close:
09-25 14:51:49.469 10662-10702/com.example.anecdot V/AbsUVCCameraHandler: close:finished
09-25 14:51:49.469 10662-10703/com.example.anecdot V/CameraThread: handleRelease:mIsRecording=false
09-25 14:51:49.469 10662-10703/com.example.anecdot V/CameraThread: handleClose:
09-25 14:51:49.469 10662-10703/com.example.anecdot V/UVCCameraTextureView: setVideoEncoder:
09-25 14:51:49.469 10662-10703/com.example.anecdot V/CameraThread: handleRelease:finished
09-25 14:51:49.470 10662-10702/com.example.anecdot V/CameraThread: getHandler:
09-25 14:51:49.470 10662-10705/com.example.anecdot V/UVCCameraTextureView: RenderThread#setEncoder:encoder=null
09-25 14:51:49.471 10662-10702/com.example.anecdot V/UVCCameraTextureView: getPreviewTexture:
09-25 14:51:49.471 10662-10705/com.example.anecdot I/UVCCameraTextureView: RenderThread#updatePreviewSurface:
09-25 14:51:49.471 10662-10705/com.example.anecdot D/UVCCameraTextureView: updatePreviewSurface:release mPreviewSurface
09-25 14:51:49.471 10662-10705/com.example.anecdot V/UVCCameraTextureView: updatePreviewSurface:tex_id=1`

This is how I try to set the preview:
`USBMonitor monitor = mCameraHelper.getUSBMonitor();
monitor.openDevice(device);
mCameraHelper.createUVCCamera();
mCameraHelper.startPreview(mUVCCameraView);`

closed 2020-10-05 by Vishalratnam

duongnxptit, 2021-02-21:
@Vishalratnam Have you solved this problem yet?

Vishalratnam, 2021-02-22:
> @Vishalratnam Have you solved this problem yet?

Yes

bandrarom, 2022-01-05:
@Vishalratnam 
can you tell us how you  do it

CQHO, 2022-03-11:
I meet the same issue there. can't preview camera on a fragment