jiangdongguo/AndroidUSBCamera #275: 1080p high resolution problem

heaveninfosoft, 2020-10-14:
1080p high resolution work some time then gray some part in screen

https://drive.google.com/file/d/1TVjQUHuTqXqg9iIAnmvXtwFsvvqTgoKU/view?usp=sharing

idr-pt, 2020-10-22:
Same problem here . But Changing to YUYV frame format its solved . But cant Capture Pics With it

heaveninfosoft, 2020-10-22:
@idr-pt what need changes code can you show me?

idr-pt, 2020-10-22:
in Oncreate add code like this

        ```
mCameraHelper?.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_YUYV)
        mCameraHelper?.initUSBMonitor(this, mUVCCameraView, listener)
        mCameraHelper?.updateResolution(1024, 768)
```

heaveninfosoft, 2020-10-22:
but then we cant get previewListernerFrame right???

idr-pt, 2020-10-22:
i think we can . but cant capture photo

wsl1015310333, 2020-11-18:
I had the same problem . Have you solved the problem？

wsl1015310333, 2020-11-24:
but  mCameraHelper?.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_YUYV)  is No video preview  at android 10 
，mCameraHelper?.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_MJPEG) can has video  preview at android 10