jiangdongguo/AndroidUSBCamera #278: UNABLE TO capture IMAGE

KingBrianZA, 2020-10-22:
when i click capture image it not saving it and the video is empty when i record

quocnhat7, 2020-10-26:
perhaps your app did not request storage permission before staring the capturer code?
You can read the code again to find the reason

burcutopcu, 2020-10-26:
I have the same issue, the video is empty when I record, did you solve it? @KingBrianZA

vp-007, 2020-10-27:
I am facing the issue with video recording. 

However solved the image capture issue solved by changing following lines in AbstractUVCCameraHandler.java file:
```
case MSG_CAPTURE_STILL:
		  thread.handleCaptureStill((String) msg.obj);
//                thread.handleStillPicture((String) msg.obj);
                break;
```

silvioalvim, 2020-12-16:
> I am facing the issue with video recording.
>
> However solved the image capture issue solved by changing following lines in AbstractUVCCameraHandler.java file:
>
> ```
> case MSG_CAPTURE_STILL: [...]

Yes, thats fix the image capture, thank you! Any news about video recording?