jiangdongguo/AndroidUSBCamera #280: video recording captures audio only

vp-007, 2020-10-27:
I'm trying to build the camera application which can capture photos as well as videos. 
Also testing it on Android 10 platform. It must run on A9 and A10 both.

When I try to capture a video,  it starts a recording and saves .mp4 file on the targeted directory. It has only audio information only not video. It doesn't contain any video information there.

When I check in H264EncoderConsumer file, the below line always returns to -1 and goes in no output available yet. 
`outputBufferIndex = mMediaCodec.dequeueOutputBuffer(bufferInfo, TIMES_OUT);`

Checked on Mp4MediaMuxer, mVideoTrackIndex == -1 and so, it returns directly without any further processing.
```
if ((!isVoiceClose && mAudioTrackIndex == -1) || mVideoTrackIndex == -1) {
            Log.i(TAG, String.format("pumpStream [%s] but muxer is not start.ignore..", isVideo ? "video" : "audio"));
            return;
}
```

As far as I understood, It is not adding video track through below function in Mp4MediaMuxer, because `isVideo `remains always `false` in below. 
```
public synchronized void addTrack(MediaFormat format, boolean isVideo) {
        // now that we have the Magic Goodies, start the muxer
        if ((!isVoiceClose && mAudioTrackIndex != -1) && mVideoTrackIndex != -1)
            throw new RuntimeException("already add all tracks");

        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.JELLY_BEAN_MR2) {
            int track = mMuxer.addTrack(format);
            if (VERBOSE)
                Log.i(TAG, String.format("addTrack %s result %d", isVideo ? "video" : "audio", track));
            if (isVideo) {
                mVideoFormat = format;
                mVideoTrackIndex = track;
                // When the audio track is added Or start mute
                if (isVoiceClose || mAudioTrackIndex != -1) {
                    if (VERBOSE)
                        Log.i(TAG, "both audio and video added,and muxer is started");
                    mMuxer.start();
                    mBeginMillis = System.currentTimeMillis();
                }
            } else {
                mAudioFormat = format;
                mAudioTrackIndex = track;
                if (mVideoTrackIndex != -1) {
                    mMuxer.start();
                    mBeginMillis = System.currentTimeMillis();
                }
            }
        }
    }
```

burcutopcu, 2020-10-29:
I have the same problem, as you said I change isVideo to true inside the code, but the result didn't change @jiangdongguo @vp-007

vp-007, 2020-11-12:
@burcutopcu I resolved it. The frame format type JPEG may not be supported in your mobile device, you can change it to YUYV format and try. 

Go to the USBCameraActivity.java file and change frame format like below.
```
        //mCameraHelper.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_MJPEG);
        mCameraHelper.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_YUYV);
```

vp-007, 2020-11-12:
@burcutopcu I resolved it. The frame format type JPEG may not be supported in your mobile device, you can change it to YUYV format and try.

Go to the USBCameraActivity.java file and change frame format like below.

        //mCameraHelper.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_MJPEG);
        mCameraHelper.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_YUYV);

closed 2020-11-12 by vp-007

burcutopcu, 2020-11-16:
@vp-007 It is working thanks!! 🥇

silvioalvim, 2020-12-16:
Fantastic, @vp-007! Thank you!

burcutopcu, 2020-12-16:
I have a problem on saving video nowadays, it was working before but today video is not saving. Do you have a similar problem? @silvioalvim @vp-007

silvioalvim, 2020-12-16:
After the modification indicated above, for me it is saving the videos correctly. Now I'm trying to work on the overlay information @burcutopcu