jiangdongguo/AndroidUSBCamera #284: Captured Video File is Empty

burcutopcu, 2020-10-29:
For video recording there is a problem that I can not understand, it saves the video as mp4 but the video looks empty. When I try to open this saved video, it says that video uri is invalid. Is there somebody else that struggles with the same issue? How can we solve this issue?

heaveninfosoft, 2020-10-29:
there is many problem you cant figure out in one line

burcutopcu, 2020-10-29:
is there any library that works properly for video capturing in uvc camera, do you have any information? I coulnd't find any library for video capturing in uvc camera except this. @heaveninfosoft

silvioalvim, 2020-12-15:
Does anyone have any news about this? Did you manage to solve this problem? I'm stuck with that too