jiangdongguo/AndroidUSBCamera #285: 【resolved】could not open camera:err=-50

LannyLee, 2020-11-07:
`if (!isRequest) {
                isRequest = true;
                if (mCameraHelper != null) {
                    mCameraHelper.requestPermission(1);// Change it to front or back（1 or 0）
                }
            }`

LannyLee, 2020-11-07:
>             if (!isRequest) {
                isRequest = true;
                if (mCameraHelper != null) {
                    mCameraHelper.requestPermission(1);// Change it to front or back（1 or 0）
                }
            }

ping3333, 2020-11-20:
@LannyLee 非常感谢！同样遇到这个问题，已解决。开发的时候使用的摄像头一直都是返回的一个device，结果测试时用的设备没发现返回了多个device，然后一直返回这个错误。看到你的解决方法才突然想起这个授权方法的参数就是device列表的下标。后面遇到的朋友请查看UVCCameraHelper类中的requestPermission方法。