jiangdongguo/AndroidUSBCamera #297: App crashes when plugging USB camera in (targetSdkVersion = 29)

Metuuu, 2020-12-10:
App crashes when plugging USB camera in.
`targetSdkVersion` is set to `29`

```
2020-12-10 16:44:24.092 10596-1673/com.jiangdg.usbcamera E/AndroidRuntime: FATAL EXCEPTION: USBMonitor
    Process: com.jiangdg.usbcamera, PID: 10596
    java.lang.SecurityException: User has not given 10543/com.jiangdg.usbcamera permission to access device /dev/bus/usb/001/002
        at android.os.Parcel.createExceptionOrNull(Parcel.java:2373)
        at android.os.Parcel.createException(Parcel.java:2357)
        at android.os.Parcel.readException(Parcel.java:2340)
        at android.os.Parcel.readException(Parcel.java:2282)
        at android.hardware.usb.IUsbSerialReader$Stub$Proxy.getSerial(IUsbSerialReader.java:123)
        at android.hardware.usb.UsbDevice.getSerialNumber(UsbDevice.java:157)
        at com.serenegiant.usb.USBMonitor.getDeviceKeyName(USBMonitor.java:733)
        at com.serenegiant.usb.USBMonitor.getDeviceKey(USBMonitor.java:766)
        at com.serenegiant.usb.USBMonitor.updatePermission(USBMonitor.java:469)
        at com.serenegiant.usb.USBMonitor.hasPermission(USBMonitor.java:459)
        at com.serenegiant.usb.USBMonitor$2.run(USBMonitor.java:596)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:245)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: android.os.RemoteException: Remote stack trace:
        at com.android.server.usb.UsbUserPermissionManager.checkPermission(UsbUserPermissionManager.java:654)
        at com.android.server.usb.UsbSerialReader.getSerial(UsbSerialReader.java:98)
        at android.hardware.usb.IUsbSerialReader$Stub.onTransact(IUsbSerialReader.java:84)
        at android.os.Binder.execTransactInternal(Binder.java:1165)
        at android.os.Binder.execTransact(Binder.java:1134)
```

Metuuu, 2020-12-11:
It seems that Android 9 and later needs CAMERA permission to access UVC devices.

I tested UVC permissions with Saki's UVCPermissionTest app and permissions were granted for my USB camera.
(targetSdkVersion was set to 29)
https://github.com/saki4510t/UVCPermissionTest

Metuuu, 2020-12-11:
Device serial number can't be read without usb device access permission when targetSdkVersion is set to higher than 28.
https://developer.android.com/reference/android/hardware/usb/UsbDevice#getSerialNumber()

[CAMERA](https://developer.android.com/reference/android/Manifest.permission#CAMERA) permission is also required when targeting sdk version > 27.

I'll do a pull request soon regarding to these issues.

Metuuu, 2020-12-11:
There is also another issue that affects some devices.
App crashes when USB camera is plugged out when targetSdkVersion > 27
https://github.com/saki4510t/UVCCamera/issues/446

ToysoftInc, 2021-01-10:
I had similar problem.  What work for me is set versionTarget = 27 in the lib gradle

closed 2021-01-15 by Metuuu

Metuuu, 2021-01-15:
I am trying to get USB cameras working for > 27 versions so switching back to 27 isn't an option.

I stopped using this wrapper library and switched using the main library https://github.com/saki4510t/UVCCamera.

evangunawan, 2021-05-31:
> I am trying to get USB cameras working for > 27 versions so switching back to 27 isn't an option.
> 
> I stopped using this wrapper library and switched using the main library https://github.com/saki4510t/UVCCamera.

Hello! got the same issues, I also want to implement it on > 27 version.
Do you succeeded with the implementation with the main library?
Thank you!

Metuuu, 2021-05-31:
> > I am trying to get USB cameras working for > 27 versions so switching back to 27 isn't an option.
> > I stopped using this wrapper library and switched using the main library https://github.com/saki4510t/UVCCamera.
> 
> @Metuuu
> Hello! got the same issue, I also want to implement it on > 27 version.
> Did you succeeded implementing the functionality with the main library?
> Thank you!

@evangunawan 
I had to fork the library and push some changes. You can check it https://github.com/Metuuu/UVCCamera and try if usbCameraTest8 works for you.

evangunawan, 2021-05-31:
Thank you for the reply!
Tried to build usbCameraTest8, but app got crashed when applying the camera (usb device read on the app).
Built and installed the debug apk to my phone, galaxy A51 Android 11.

Metuuu, 2021-05-31:
Would you mind creating issue to the repo and provide the error stack. I might add comment to it but I don't promise to fix anything 🙂
I merged changes from the original repo to mine so you can try if they fixed anything. You could also try test 7 and 6 and check if they fail with same error.

evangunawan, 2021-05-31:
Gonna check soon! I'll catch up on your repo issue later! thank you very much 😊

evangunawan, 2021-06-02:
Seems that the error is my fault configuring the project.
It worked and ported into my application, I just need to do some layout styling now lol. thank you very much! @Metuuu

heaveninfosoft, 2021-12-11:
@Metuuu @evangunawan hello brother can you help me i nned to access usb camera >=30 target i tried many lib all crash
https://github.com/Metuuu/UVCCamera  this not crash but after usb permison camera name show but preview at all please please help

ToysoftInc, 2021-12-11:
@heaveninfosoft I included the usb lib source in my project and the project gradle

`ext {
    androidXVersion = '1.1.0'  // variable that can be referenced to keep support libs consistent
    commonLibVersion= '4.1.1'
    versionCompiler = 28
    // when target=27, 9.0 is ok
    versionTarget = 27
    // if hope supporting 4.4
    // please modify it to 16
    minSdkVersion = 21
    versionCode = 99
    versionNameString = '1.2.1.20200414'
    javaSourceCompatibility = JavaVersion.VERSION_1_8
    javaTargetCompatibility = JavaVersion.VERSION_1_8
}`

It is working for me on Android 10+

heaveninfosoft, 2021-12-11:
Bro i am taking about targetSdkVersion = 29 not targetSdkVersion = 27.

targetSdkVersion = 27 work fine but targetSdkVersion = 29 crash

ToysoftInc, 2021-12-11:
oh. My bad.  I think I had problem with 29.  That is why I stick with 27 :)

heaveninfosoft, 2021-12-11:
No problem, thanks!
I need solutions for targetSdkVersion = 29