jiangdongguo/AndroidUSBCamera #302: Green lines at the bottom in recorded video

ToysoftInc, 2021-01-10:
I'm about to record video but at the bottom of the video has several green lines.
why is this?  I'm running the app on Samsung Note 9

            mUSBCameraHelper.setDefaultFrameFormat(UVCCamera.PIXEL_FORMAT_NV21);

Other format will not work.

thanks

silvioalvim, 2021-02-04:
`mCameraHelper.setDefaultFrameFormat(UVCCameraHelper.FRAME_FORMAT_YUYV);`

That above worked for me

g-mu, 2021-02-28:
I have this same issue - green bar and blue color shift. Changing the frame format doesn't change the recorded video. The preview looks fine and still photos look fine, just video recordings are wrong. I've tried two different cameras. Does anyone have a different solution I can try? Thanks.

g-mu, 2021-02-28:
Edit: I've found that the green bar is caused by na offset of the U and V channels by X number of rows. The X number of rows seems to be dependent on resolution and maybe some frame size formula. I haven't found a formula for determining the number of rows to pad yet. I've written a hack in H264EncodeConsumer.java for the moment. Hope it helps someone else.

    // 640x480: 0 rows padding
    // 1280x720: 16 rows padding
    // 1920x1080: 8 rows padding
    // YYYYYYYY UVUV(nv12)--> YYYYYYYY VUVU(nv21)
  private byte[] nv12ToNV21_G(byte[] nv12, int width, int height) {
        int yPaddingRows = 0;
        if(width == 1280){
            yPaddingRows = 16;
        } else if(width == 1920){
            yPaddingRows = 8;
        }
        int subComponentOffset = yPaddingRows * width;
        byte[] ret = new byte[nv12.length+subComponentOffset];
        int framesize = width * height;
        int i = 0, j = 0;
        // copy Y component
        System.arraycopy(nv12, 0, ret, 0, framesize); // green
        // copy U and V components
        for (j = framesize; j < nv12.length; j += 2) {
            ret[j + subComponentOffset] = nv12[j+1];
            ret[j+1 + subComponentOffset] = nv12[j];
        }
        return ret;
    }

closed 2021-04-25 by jiangdongguo

Clamorae, 2022-11-30:
I have the same problem where do you call your method?