jiangdongguo/AndroidUSBCamera #307: the preview image is broken as a gray

dudgus0627, 2021-02-04:
When returning after switching to another app, the preview image is broken as a gray.
I tried to get camera images on "onFrame" IFrameCallback (AbstractUVCCameraHandler.java), I've also got the same broken images.
Maybe it occurred on NativeCode.
Could you check this issue?

jiangdongguo, 2022-07-13:
Please update to implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.1.7', and have a try

closed 2022-07-13 by jiangdongguo

am3n, 2023-01-30:
In version 3.1.8, sometimes the frame rate reaches the minimum possible and the image becomes completely gray

jiangdongguo, 2023-02-15:
how about 3.3.3？

jiangdongguo, 2023-02-15:
update to new of master and have a try

am3n, 2023-02-15:
I have updated to the latest commit, the situation is better than before and the problem occurs less, thanks for that.
But because the problem is not completely solved, it is not possible to use it in production mode.
I don't know exactly where the problem is, I was trying to replace MediaCodec with Surface and GLES and convert H264 directly to YUV and then to Bitmap, but it didn't work. It seems that the UVC library does not work without Surface Preview.

jiangdongguo, 2023-02-15:
Can you show me the record video?

am3n, 2023-02-15:
https://drive.google.com/file/d/1cBUKC9Nm4yEtA0UpWfSNuHVZRsjvpPRG/view?usp=sharing

BruceZhang2017, 2024-10-31:
@jiangdongguo 这个问题还是存在的3.3.3。