jiangdongguo/AndroidUSBCamera #312: android调用uvc usb摄像头录像，但是录出来的视频是绿色，有知道原因的吗？

zhangze123456, 2021-03-15:
大佬们，android调用uvc usb摄像头录像，但是录出来的视频是绿色，有知道原因的吗？ 华为手机预览和成像正常，小米手机预览没有问题，录像后的视频是绿色的，有清楚原因的吗？
https://zxyj-stor-test-01.oss-cn-shanghai.aliyuncs.com/2021/03/12/3020/video_20210312_101703.mp4   小米手机插上外置摄像头绿屏的视频文件
https://zxyj-stor-test-01.oss-cn-shanghai.aliyuncs.com/2021/03/12/3020/20210312_101531.mp4    小米手机没有插上摄像头 正常的视频问题

jiangdongguo, 2021-03-16:
颜色格式不匹配

zhangze123456, 2021-04-01:
您好！之前的小米手机 蓝色的问题解决了，解决方法是（H264EncodeConsumer.java类 隐藏  MediaCodecInfo.CodecCapabilities.COLOR_QCOM_FormatYUV420SemiPlanar,）现在又换了一台外置usb，现在预览是红色的，请问，可以从哪里跟进，而不影响以前的手机型号和外置usb的适配， 测试手机华为P30 PRO 现在现象视频地址 ：https://zxyj-stor-test-01.oss-cn-shanghai.aliyuncs.com/2021/04/01/0/video_20210401_100547.mp4   （人物穿的衣服是黑色 衣服，预览就是红色了）

closed 2021-04-25 by jiangdongguo

chenxiaofei910714, 2021-12-14:
如何解视频绿色现象呢MediaCodecInfo.CodecCapabilities.COLOR_QCOM_FormatYUV420SemiPlanar隐藏并没什么作用