jiangdongguo/AndroidUSBCamera #319: Crash on detach usb camera

MassCraxx, 2021-03-25:
Whenever a connected usb cam is detached while the video is active, the application crashes. This was tested on an Android 11 device with your sample app main branch running with targetSDK 27. In this case, it produces a SIGSEGV with a null pointer dereference cause, I have also seen SIGABRT with Abort message: 'FORTIFY: pthread_mutex_lock called on a destroyed mutex (0x6e369cb4f4)', both are produced at similar spots in the c code, while handling the usb event.

2021-03-25 17:20:50.709 1149-1536/? D/UsbHostManager: usbDeviceRemoved - start: deviceAddress=/dev/bus/usb/001/003
2021-03-25 17:20:50.709 1149-1536/? D/UsbHostManager: Removed device at /dev/bus/usb/001/003 was already gone
2021-03-25 17:20:50.710 1149-1536/? D/UsbHostManager: usbDeviceRemoved - end
2021-03-25 17:20:50.725 15071-15071/? I/crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstone
2021-03-25 17:20:50.725 664-664/? I/tombstoned: received crash request for pid 15048
2021-03-25 17:20:50.726 15071-15071/? I/crash_dump64: performing dump of process 14807 (target tid = 15048)
2021-03-25 17:20:50.730 15071-15071/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2021-03-25 17:20:50.730 15071-15071/? A/DEBUG: Build fingerprint: 'samsung/x1sxeea/x1s:11/RP1A.200720.012/G981BXXS6DUBA:user/release-keys'
2021-03-25 17:20:50.730 15071-15071/? A/DEBUG: Revision: '22'
2021-03-25 17:20:50.730 15071-15071/? A/DEBUG: ABI: 'arm64'
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG: Timestamp: 2021-03-25 17:20:50+0100
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG: pid: 14807, tid: 15048, name: CameraThread  >>> com.jiangdg.usbcamera <<<
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG: uid: 10343
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG: Cause: null pointer dereference
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG:     x0  0000000000000010  x1  000000704a3768c5  x2  0000006d4596a761  x3  0000006d4596a78f
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG:     x4  0000000000000001  x5  00000000ffffff94  x6  0000000000000000  x7  00000000004ee2e4
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG:     x8  0000000000000000  x9  0000000000000010  x10 0000006d458bd9c0  x11 0000006d458bda30
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG:     x12 ffffff80ffffffd0  x13 00000000605cb862  x14 00263bd188b53f64  x15 000039f8bec43a38
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG:     x16 0000007049bcf170  x17 000000704e346434  x18 0000006d45764000  x19 000000000000000c
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG:     x20 0000006e369502b4  x21 0000006de69318d0  x22 0000000000000004  x23 0000000000000004
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG:     x24 0000006e369504a0  x25 0000000000000000  x26 0000000000000002  x27 0000006db6925610
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG:     x28 0000000000000000  x29 0000006d458bdaf0
2021-03-25 17:20:50.731 15071-15071/? A/DEBUG:     lr  0000006d45964280  sp  0000006d458bda30  pc  0000006d45964290  pst 0000000080001000
2021-03-25 17:20:50.732 15071-15071/? A/DEBUG: backtrace:
2021-03-25 17:20:50.732 1149-1303/? D/UsbUI: onUEvent(Host Interface): {SUBSYSTEM=usb, SEQNUM=14986, ACTION=unbind, INTERFACE=9/0/2, DEVTYPE=usb_interface, PRODUCT=1a40/201/100, DEVPATH=/devices/platform/10e00000.usb/10e00000.dwc3/xhci-hcd.12.auto/usb1/1-1/1-1:1.0, TYPE=9/0/2}
2021-03-25 17:20:50.732 15071-15071/? A/DEBUG:       #00 pc 0000000000010290  /data/app/~~Z4CxqiiyIjc-T2Uf45NkDQ==/com.jiangdg.usbcamera-qEUuSZ-pwu1OgV4HbUqW_Q==/lib/arm64/libusb100.so (BuildId: 23d7dd0bcf4c20694d9b4c2aa3572064e47f2f34)
2021-03-25 17:20:50.732 15071-15071/? A/DEBUG:       #01 pc 000000000000c28c  /data/app/~~Z4CxqiiyIjc-T2Uf45NkDQ==/com.jiangdg.usbcamera-qEUuSZ-pwu1OgV4HbUqW_Q==/lib/arm64/libusb100.so (BuildId: 23d7dd0bcf4c20694d9b4c2aa3572064e47f2f34)
2021-03-25 17:20:50.732 15071-15071/? A/DEBUG:       #02 pc 000000000000bcf4  /data/app/~~Z4CxqiiyIjc-T2Uf45NkDQ==/com.jiangdg.usbcamera-qEUuSZ-pwu1OgV4HbUqW_Q==/lib/arm64/libusb100.so (libusb_handle_events_timeout_completed+748) (BuildId: 23d7dd0bcf4c20694d9b4c2aa3572064e47f2f34)
2021-03-25 17:20:50.732 15071-15071/? A/DEBUG:       #03 pc 000000000000c414  /data/app/~~Z4CxqiiyIjc-T2Uf45NkDQ==/com.jiangdg.usbcamera-qEUuSZ-pwu1OgV4HbUqW_Q==/lib/arm64/libusb100.so (libusb_handle_events+44) (BuildId: 23d7dd0bcf4c20694d9b4c2aa3572064e47f2f34)
2021-03-25 17:20:50.732 15071-15071/? A/DEBUG:       #04 pc 0000000000011e24  /data/app/~~Z4CxqiiyIjc-T2Uf45NkDQ==/com.jiangdg.usbcamera-qEUuSZ-pwu1OgV4HbUqW_Q==/lib/arm64/libuvc.so (_uvc_handle_events+132) (BuildId: abac8c7f85e647250fdf0142ada1bec4e59c2d74)
2021-03-25 17:20:50.732 15071-15071/? A/DEBUG:       #05 pc 00000000000b0228  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+64) (BuildId: 0537c1e6838792775ce275f0ee9e5270)
2021-03-25 17:20:50.732 15071-15071/? A/DEBUG:       #06 pc 000000000005070c  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: 0537c1e6838792775ce275f0ee9e5270)
2021-03-25 17:20:50.733 1149-1303/? D/UsbUI: onUEvent(Host Interface): {SUBSYSTEM=usb, SEQNUM=14987, ACTION=remove, INTERFACE=9/0/2, DEVTYPE=usb_interface, PRODUCT=1a40/201/100, MODALIAS=usb:v1A40p0201d0100dc09dsc00dp02ic09isc00ip02in00, DEVPATH=/devices/platform/10e00000.usb/10e00000.dwc3/xhci-hcd.12.auto/usb1/1-1/1-1:1.0, TYPE=9/0/2}
2021-03-25 17:20:50.733 1149-1303/? D/UsbUI: notifyUsbInterface: [9, 0, 2]
2021-03-25 17:20:50.738 1149-1536/? D/UsbHostManager: usbDeviceRemoved - start: deviceAddress=/dev/bus/usb/001/002
2021-03-25 17:20:50.738 1149-1536/? D/UsbHostManager: Removed device at /dev/bus/usb/001/002 was already gone
2021-03-25 17:20:50.739 1149-1536/? D/UsbHostManager: usbDeviceRemoved - end

jiangdongguo, 2021-04-08:
which version 2.3.5.?

MassCraxx, 2021-04-08:
Yes, this was on 2.3.5

closed 2021-04-25 by jiangdongguo

LeiStoneLi, 2021-04-27:
@MassCraxx Did you test this on 2.3.6? Looking forward to your reply！Thank you!

MassCraxx, 2021-04-30:
Seems to be working fine in your sample app now, thanks for the update!