jiangdongguo/AndroidUSBCamera #324: APP caused the phone system to crash and restart.

LeiStoneLi, 2021-05-14:
Phont Type：Google pixel4
Sytetem：Android11
updated to the latest patch,(2021.5)

APP caused the phone system to crash and restart.

LeiStoneLi, 2021-06-15:
#00 pc 00000000000132cc /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) [arm64-v8a::95f68a34fdf7b31f42d3f811d5d88f58]
#01 pc 0000000000076428 /apex/com.android.runtime/lib64/bionic/libc.so (_res_stats_usable_server+84) [arm64-v8a::95f68a34fdf7b31f42d3f811d5d88f58]
#02 pc 0000000000075a24 /apex/com.android.runtime/lib64/bionic/libc.so (__res_nsend+6252) [arm64-v8a::95f68a34fdf7b31f42d3f811d5d88f58]
#03 pc 000000000007587c /apex/com.android.runtime/lib64/bionic/libc.so (pthread_mutex_lock+160) [arm64-v8a::95f68a34fdf7b31f42d3f811d5d88f58]
#04 pc 0000000000010208  /data/app/com.jiangdg.usbcamera-wD53yFl2fd1vquCYnjf5Sg==/lib/arm64/libusb100.so [arm64-v8a::4e295f7658d8c85a9c614c0626b535cd]
#05 pc 000000000000c28c  /data/app/com.jiangdg.usbcamera-wD53yFl2fd1vquCYnjf5Sg==/lib/arm64/libusb100.so [arm64-v8a::4e295f7658d8c85a9c614c0626b535cd]
#06 pc 000000000000bcf4  /data/app/com.jiangdg.usbcamera-wD53yFl2fd1vquCYnjf5Sg==/lib/arm64/libusb100.so (libusb_handle_events_timeout_completed+748) [arm64-v8a::4e295f7658d8c85a9c614c0626b535cd]
#07 pc 000000000000c414  /data/app/com.jiangdg.usbcamera-wD53yFl2fd1vquCYnjf5Sg==/lib/arm64/libusb100.so (libusb_handle_events+44) [arm64-v8a::4e295f7658d8c85a9c614c0626b535cd]
#08 pc 0000000000011e24  /data/app/com.jiangdg.usbcamera-wD53yFl2fd1vquCYnjf5Sg==/lib/arm64/libuvc.so (_uvc_handle_events+132) [arm64-v8a::65914680ce546556f4819427a6cc0cfd]
#09 pc 0000000000074d4c /apex/com.android.runtime/lib64/bionic/libc.so (__res_nsend+2964) [arm64-v8a::95f68a34fdf7b31f42d3f811d5d88f58]
#10 pc 0000000000015288 /apex/com.android.runtime/lib64/bionic/libc.so [arm64-v8a::95f68a34fdf7b31f42d3f811d5d88f58]

mking123456, 2022-03-31:
Have you found the reason？ I runned demo on RK3288 board, it caused system crash and restart too.

closed 2022-07-13 by jiangdongguo