jiangdongguo/AndroidUSBCamera #330: It is not working on Android version 10

KanwalKhan-20, 2021-06-18:
It is connecting but preview is unavailable

ltrhao-wxh, 2021-08-11:
Yes, I have the same problem.

ltrhao-wxh, 2021-08-23:
Is there any solution?

EworkitSrl, 2021-11-15:
have you solve it

ashrafulislamdolon, 2021-11-26:
**@EworkitSrl**  facing the same problem..have you solved it yet??

KanwalKhan-20, 2021-11-27:
Not yet
I have implemented another library

ashrafulislamdolon, 2021-11-28:
@KanwalKhan-20 can u please share the details?

ashrafulislamdolon, 2021-12-02:
@KanwalKhan-20 please share the details about the library

KanwalKhan-20, 2021-12-07:
I have tested another library on updated versions and it is not working on
Android 11
Sorry for the inconvenience.

JulesD29, 2022-06-08:
Any news here ?

jiangdongguo, 2022-07-13:
Please update to implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.1.7', and have a try

closed 2022-07-13 by jiangdongguo

KanwalKhan-20, 2022-09-12:
Hi @jiangdongguo  
Where to add this line.
Because after adding it in gradle I am getting duplicate error.

JulesD29, 2022-09-12:
Hi @KanwalKhan-20, I had to unload old module to make it work and to comment the part :

    implementation("com.serenegiant:common:${commonLibVersion}") {
        exclude module: 'support-v4'
    }

I don't know if it's a good thing, @jiangdongguo is this the right way ?

KanwalKhan-20, 2022-09-12:
Hi @JulesD29 
[image]
If I am doing this then where I need to add
 implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.1.7'

Thanks for your support

KanwalKhan-20, 2022-09-12:
> Hi @KanwalKhan-20, I had to unload old module to make it work and to comment the part :
> 
> ```
> implementation("com.serenegiant:common:${commonLibVersion}") {
>     exclude module: 'support-v4'
> }
> ```
> 
> I don't know if it's a good thing, @jiangdongguo is this the right way ?

[image]

Getting an error

JulesD29, 2022-09-12:
Do you imported the github of @jiangdongguo like this ?
    implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.2.7'

And do you add this in your build.gradle(:app) ?
repositories {
    allprojects {
        repositories {
            google()
            jcenter()
            maven { url "https://jitpack.io" }
        }
    }
}

KanwalKhan-20, 2022-09-12:
I have imported the library.

KanwalKhan-20, 2022-09-12:
> > Do you imported the github of @jiangdongguo like this (in your build.gradle(:app) in dependencies ? implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.2.7'
> > And do you add this in your build.gradle(:app) ? repositories { allprojects { repositories { google() jcenter() maven { url "https://jitpack.io" } } } }
> 
> No

[image]

[image]

JulesD29, 2022-09-12:
OK, I don't know, but in the ReadMe of @jiangdongguo it says that you have to put these lines :

allprojects {
    repositories {
        google()
        jcenter()
        maven { url "https://jitpack.io" }
    }
}

https://github.com/jiangdongguo/AndroidUSBCamera#:~:text=allprojects%20%7B%0A%20%20%20%20repositories%20%7B%0A%20%20%20%20%20%20%20%20google()%0A%20%20%20%20%20%20%20%20jcenter()%0A%20%20%20%20%20%20%20%20maven%20%7B%20url%20%22https%3A//jitpack.io%22%20%7D%0A%20%20%20%20%7D%0A%7D