jiangdongguo/AndroidUSBCamera #337: 连接了两个USB摄像头，可以单独预览，不能同时预览

Jack-Hu, 2021-07-06:
核心日志如下
W/libUVCCamera: [2419*UVCPreview.cpp:119:get_frame]:allocate new frame
W/libUVCCamera: [2418*UVCPreview.cpp:119:get_frame]:allocate new frame
E/BufferQueueProducer: [SurfaceTexture-1-2378-2] connect: BufferQueue has been abandoned
E/BufferQueueProducer: [SurfaceTexture-1-2378-2] connect: BufferQueue has been abandoned
E/BufferQueueProducer: [SurfaceTexture-1-2378-2] connect: BufferQueue has been abandoned
E/libuvc/stream: [2423*stream.c:1605:uvc_stream_start_bandwidth]:fail
W/libUVCCamera: [2423*diag.c:88:uvc_perror]:failed start_streaming:I/O error (-1)
E/BufferQueueProducer: [SurfaceTexture-1-2378-2] connect: BufferQueue has been abandoned
E/BufferQueueProducer: [SurfaceTexture-1-2378-2] connect: BufferQueue has been abandoned
E/BufferQueueProducer: [SurfaceTexture-1-2378-2] connect: BufferQueue has been abandoned
...
//后面都是 E/BufferQueueProducer: [SurfaceTexture-1-2378-2] connect: BufferQueue has been abandoned

摄像头的参数，USB带宽等，也不清楚。
在#62中，用leessy的方案就是以上的日志。
望大神指点

sheajin, 2021-08-02:
我也遇到了同样的问题，请问你解决了吗？

Jack-Hu, 2021-08-06:
换方案了

chenYuCui, 2022-05-09:
> 换方案了
> […](#)

你好，能否转发一下邮件，我也有相同的问题

chenYuCui, 2022-05-09:
> 我也遇到了同样的问题，请问你解决了吗？

你解决了吗？

sheajin, 2022-05-11:
> > 我也遇到了同样的问题，请问你解决了吗？
> 
> 你解决了吗？

直接用UVCCamera显示的