jiangdongguo/AndroidUSBCamera #343: take picture and record video do not work

HM-Houda, 2021-08-12:
Hi everyone,
I've just cloned the projet and compiled it.
I generated a signed apk, however after installation on smartphone with Android 9.
I can view camera but the options of take picture and recording don't work.

Any help would be appreciated.
Thank you very much

ashrafulislamdolon, 2021-11-26:
> 

can u please tell me how to work with this code in a higher version(11/12) of android?
The app does open on an android 11 device but the screen remains black.

deebashd, 2021-11-26:
@ashrafulislamdolon Right now Android 11 and 12 is not supporting requestPermission method, so it won't work when your targetsdk is 30 and above. Set your targetsdk to less than 29, requestPermission will work. 

Note: If your targetsdk is less than 30, you can't update or deploy your app in Google Playstore from November 1, 2021. If you really want to give USB camera feature to your users, you have to ask them download from your website not from Playstore.

In Android forum, they mentioned about deploying fix in Android 12, but it's not rolled out yet.

ashrafulislamdolon, 2021-11-26:
@deebashd with the current setting, the app opens in android 11 devices and it also connects to the camera but the screen remains black..how to solve this problem??
please help me

ashrafulislamdolon, 2021-12-02:
@deebashd hello..can u help me please??

ashrafulislamdolon, 2021-12-02:
@HM-Houda the app opens in android 11 but no preview is available..have you faced this problem??

deebashd, 2021-12-02:
@ashrafulislamdolon App will open without any problem, but you can't request permission when your app targets SDK 30 and above. Without permission, nothing will work. You can decrease the target SDK but you can't update the app to PlayStore instead you can make your user download APK directly from your website.

ashrafulislamdolon, 2021-12-02:
@deebashd this is the current setting-
androidXVersion = '1.1.0'
    versionCompiler = 28
    // when target=27, 9.0 is ok
    versionTarget = 27
    // if hope supporting 4.4
    // please modify it to 16
    minSdkVersion = 16
    versionCode = 101
    versionNameString = '2.3.6.20211117'
    javaSourceCompatibility = JavaVersion.VERSION_1_8
    javaTargetCompatibility = JavaVersion.VERSION_1_8

I can install and open this app in android 11 device but the camera preview is unavailable..

deebashd, 2021-12-02:
When you connect the USB camera to your phone running Android 11, does the phone pops up Permission request dialog automatically? And it should open your app automatically too.
1. If it is happening and still you're getting black screen after giving permission, there might be problem in your code.
2. If no permission dialog pops up then that's the same problem I mentioned above.

And also please share your manifest.

ashrafulislamdolon, 2021-12-02:
@deebashd yes, a permission request pops up when I connect the camera.

here is the manifest file-

<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.jiangdg.usbcamera">

    <uses-permission android:name="android.permission.RECORD_AUDIO"/>
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />

    <uses-feature android:name="android.hardware.usb.host"/>

    <application
        android:name=".application.MyApplication"
        android:allowBackup="true"
        android:icon="@mipmap/ic_launcher"
        android:label="@string/app_name"
        android:supportsRtl="true"
        android:theme="@style/AppTheme">
        <activity android:name="com.jiangdg.usbcamera.view.SplashActivity"
            android:screenOrientation="portrait">
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />

                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>
        </activity>

        <activity android:name=".view.USBCameraActivity"
            android:screenOrientation="portrait">
            <intent-filter>
                <action android:name="android.intent.action.VIEW"/>
            </intent-filter>
        </activity>
    </application>

</manifest>

ashrafulislamdolon, 2021-12-02:
I am using the same code given here-

https://github.com/jiangdongguo/AndroidUSBCamera

deebashd, 2021-12-02:
<activity android:name=".camera.MainActivity">
            <intent-filter>
                <action android:name="android.hardware.usb.action.USB_ACCESSORY_ATTACHED" />
                <action android:name="android.hardware.usb.action.USB_ACCESSORY_DETACHED" />
            </intent-filter>
            <meta-data
                android:name="android.hardware.usb.action.USB_ACCESSORY_ATTACHED"
                android:resource="@xml/accessory_filter" />
</activity>

Add the above lines in your manifest and replace the activity name. This will open up your app when USB camera is connected to your Phone. Uninstall if any other USB camera apps in your phone.

You can remove meta-data of accessory_filter if you don't know the manufacturer or model related data.
**accessory_filter.xml**

<resources>
    <usb-accessory manufacturer="vonchenchen" model="android.usbaoa" version="0.1"></usb-accessory>
</resources>

ashrafulislamdolon, 2021-12-02:
@deebashd which above lines??

deebashd, 2021-12-02:
> @deebashd which above lines??

I edited it, please recheck my above comment

ashrafulislamdolon, 2021-12-05:
hello @deebashd 

I tried what u said, but still facing the same old problem.
the app shows no preview of the camera even though it connects. I've added the screenshots, please have a look.
[image]
[image]
[image]
s

deebashd, 2021-12-06:
Sorry, I can't suggest further, the problem might be in your code. Please ask this in StackOverflow.

ashrafulislamdolon, 2021-12-12:
@deebashd can you share your full code please?

ashrafulislamdolon, 2021-12-12:
@deebashd maybe I found the problem..the code works with the devices that have Level 3 camera2 API support
can u tell me how to make it work with Full/Limited Camera2 API support?

deebashd, 2021-12-13:
@ashrafulislamdolon Sorry, I can't share the code. Please ask code related question in StackOverflow

ashrafulislamdolon, 2021-12-13:
@deebashd it's alright.
I just need the image capturing feature of this app..how to remove other menu options completely?

closed 2022-07-01 by jiangdongguo