jiangdongguo/AndroidUSBCamera #355: "Missing required attribute: dependency groupId"

cihangoksu, 2021-11-17:
Thanks a lot for the new commit 2.3.7! 
When I tried to compile it via Buildozer, I got the following error. Would there be a solution for that?

> Task :javaPreCompileDebug FAILED
[DEBUG]:   	
[DEBUG]:   esol	FAILURE: Build failed with an exception.
[DEBUG]:   	
[DEBUG]:   	* What went wrong:
[DEBUG]:   	Execution failed for task ':javaPreCompileDebug'.
[DEBUG]:   	> Could not resolve all files for configuration ':debugCompileClasspath'.
[DEBUG]:   	   > Could not resolve com.github.jiangdongguo:AndroidUSBCamera:2.3.7.
[DEBUG]:   	     Required by:
[DEBUG]:   	         project :
[DEBUG]:   	      > Could not resolve com.github.jiangdongguo:AndroidUSBCamera:2.3.7.
[DEBUG]:   	         > Could not parse POM https://jitpack.io/com/github/jiangdongguo/AndroidUSBCamera/2.3.7/AndroidUSBCamera-2.3.7.pom
[DEBUG]:   	            > Missing required attribute: dependency groupId
[DEBUG]:   	
[DEBUG]:   	* Try:
[DEBUG]:   	Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
[DEBUG]:   	
[DEBUG]:   	* Get more help at https://help.gradle.org
[DEBUG]:   	
[DEBUG]:   	Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
[DEBUG]:   	Use '--warning-mode all' to show the individual deprecation warnings.
[DEBUG]:   	See https://docs.gradle.org/6.4.1/userguide/command_line_interface.html#sec:command_line_warnings
[DEBUG]:   	
[DEBUG]:   	BUILD FAILED in 823ms
[DEBUG]:   	3 actionable tasks: 3 executed
[DEBUG]:   	
[DEBUG]:

zhuxiaoxue, 2021-11-24:
same problem

deepakrathi, 2022-01-03:
Facing the same issue? Has anybody resolved it?

AlexPetroPE, 2022-01-22:
Also having this issue.. any help?

xiaodi68tmc, 2022-03-09:
I met the same issue. Have anyone got solution?

Derek851029, 2022-03-10:
Add this to app build.grade
configurations {
        all*.exclude group: 'com.google.guava', module: 'listenablefuture'
    }