jiangdongguo/AndroidUSBCamera #358: 分段录制测试失败

wesoho, 2021-11-22:
最新版，测试分段录制，设置params.setRecordDuration(1);，实际录制得到1分钟的录像，持续录制，没有产生后续文件。

wesoho, 2021-11-22:
找到原因了：    libusbcamera\src\main\java\com\serenegiant\usb\encoder\biz\Mp4MediaMuxer.java
这个文件中定义了一个全局变量 String mFilePath;
然后在Mp4MediaMuxer中又定义了一个局部变量
public Mp4MediaMuxer(String path, long durationMillis,boolean isVoiceClose) {
        String mFilePath;
导致mFilePath为空
    public synchronized void pumpStream(ByteBuffer outputBuffer, MediaCodec.BufferInfo bufferInfo, boolean isVideo) {
中执行mMuxer = new MediaMuxer();失败

wesoho, 2021-11-22:
修改后的文件libusbcamera\src\main\java\com\serenegiant\usb\encoder\biz\Mp4MediaMuxer.java

package com.serenegiant.usb.encoder.biz;

import android.media.MediaCodec;
import android.media.MediaFormat;
import android.media.MediaMuxer;
import android.os.Build;
import android.util.Log;

import java.io.File;
import java.io.IOException;
import java.nio.ByteBuffer;

/**Mp4封装混合器
 *
 * Created by jianddongguo on 2017/7/28.
 */

public class Mp4MediaMuxer {
    private static final boolean VERBOSE = false;
    private static final String TAG = Mp4MediaMuxer.class.getSimpleName();
    private String mFilePath;
    private String mTheFilePath;
    private MediaMuxer mMuxer;
    private long durationMillis;
    private int index = 0;
    private int mVideoTrackIndex = -1;
    private int mAudioTrackIndex = -1;
    private long mBeginMillis;
    private MediaFormat mVideoFormat;
    private MediaFormat mAudioFormat;
    private boolean isVoiceClose;

    // 文件路径；文件时长
    public Mp4MediaMuxer(String path, long durationMillis,boolean isVoiceClose) {
        String mFilePath;
        mTheFilePath=path;
        this.isVoiceClose = isVoiceClose;
        this.durationMillis = durationMillis;
        if(durationMillis != 0) {
//            mFilePath = path + "-" + index++ + ".mp4";
            mFilePath = path + "-" + index + ".mp4";
            Log.i(TAG, "Mp4MediaMuxer: index="+index);
        }else{
            mFilePath = path+".mp4";
        }
        Object mux = null;
        try {
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.JELLY_BEAN_MR2) {
                mux = new MediaMuxer(mFilePath, MediaMuxer.OutputFormat.MUXER_OUTPUT_MPEG_4);
            }
        } catch (IOException e) {
            e.printStackTrace();
        } finally {
            mMuxer = (MediaMuxer) mux;
        }
    }

    public synchronized void addTrack(MediaFormat format, boolean isVideo) {
        // now that we have the Magic Goodies, start the muxer
        if ((!isVoiceClose && mAudioTrackIndex != -1) && mVideoTrackIndex != -1)
            throw new RuntimeException("already add all tracks");

        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.JELLY_BEAN_MR2) {
            int track = mMuxer.addTrack(format);
            if (VERBOSE)
                Log.i(TAG, String.format("addTrack %s result %d", isVideo ? "video" : "audio", track));
            if (isVideo) {
                mVideoFormat = format;
[... 76 lines cut ...]
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.JELLY_BEAN_MR2) {
            if (mMuxer != null) {
                //(!isVoiceClose&&mAudioTrackIndex != -1)
                if (mVideoTrackIndex  != -1) {
                    if (VERBOSE)
                        Log.i(TAG, String.format("muxer is started. now it will be stoped."));
                    try {
                        mMuxer.stop();
                        mMuxer.release();
                    } catch (IllegalStateException ex) {
                        ex.printStackTrace();
                    }

                    if (System.currentTimeMillis() - mBeginMillis <= 1500){
                        new File(mFilePath + "-" + index + ".mp4").delete();
                    }
                    mAudioTrackIndex = mVideoTrackIndex = -1;
                }else{
                    if (VERBOSE)
                        Log.i(TAG, String.format("muxer is failed to be stoped."));
                }
            }
        }
    }
}

androidscx, 2021-12-17:
非常棒，修改的地方和我猜想的一样，这个主要是解决了分段录制视频无效，永远只会生成一个视频的问题。顶上去，可以让作者修改到项目里面去

wesoho, 2021-12-17:
> 非常棒，修改的地方和我猜想的一样，这个主要是解决了分段录制视频无效，永远只会生成一个视频的问题。顶上去，可以让作者修改到项目里面去

希望能更新到项目里

androidscx, 2021-12-22:
又产生一个新问题，分段录制的时候，会多产生一个视频文件，这个视频文件是重复了第一个分段视频

wesoho, 2021-12-22:
> 

好像没遇到这个问题

androidscx, 2021-12-22:
也许是你改了吧，按道理来说视频会生成xxx-0.mp4,xxx-1.mp4,xxx-2.mp4 这样类推下去，但是我里面还会自动产生一个xxx.mp4,一脸蒙 看了半天不知道改哪里？顺便一提 System.currentTimeMillis() - mBeginMillis <= 1500 删除文件的时候也有成员变量为null的情况，稍微改下就行

closed 2022-07-01 by jiangdongguo