jiangdongguo/AndroidUSBCamera #363: crash permission to access device

Allan-Nava, 2021-12-06:
crash permission to access device
```
 java.lang.SecurityException: User has not given 10146/ permission to access device /dev/bus/usb/002/003
        at android.os.Parcel.createException(Parcel.java:2071)
        at android.os.Parcel.readException(Parcel.java:2039)
        at android.os.Parcel.readException(Parcel.java:1987)
        at android.hardware.usb.IUsbSerialReader$Stub$Proxy.getSerial(IUsbSerialReader.java:123)
        at android.hardware.usb.UsbDevice.getSerialNumber(UsbDevice.java:143)
        at com.serenegiant.usb.USBMonitor.getDeviceKeyName(USBMonitor.java:733)
        at com.serenegiant.usb.USBMonitor.getDeviceKey(USBMonitor.java:766)
        at com.serenegiant.usb.USBMonitor.updatePermission(USBMonitor.java:469)
        at com.serenegiant.usb.USBMonitor.hasPermission(USBMonitor.java:459)
        at com.serenegiant.usb.USBMonitor$2.run(USBMonitor.java:596)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)
```

heaveninfosoft, 2021-12-07:
@Allan-Nava  on which device you worked???

Allan-Nava, 2021-12-07:
> @Allan-Nava on which device you worked???

fixed it was my bad

closed 2021-12-07 by Allan-Nava

heaveninfosoft, 2021-12-08:
@Allan-Nava brother even i have same problem please tell me how you fixed bro?

Allan-Nava, 2021-12-09:
I included the library locally

heaveninfosoft, 2021-12-10:
> I included the library locally

@Allan-Nava my question what is your target version ? i interested if your target version >=30 then i need your help please

Allan-Nava, 2021-12-10:
> > I included the library locally
> 
> @Allan-Nava my question what is your target version ? i interested if your target version >=30 then i need your help please

Nope

```
ext {
    androidXVersion = '1.1.0'
    versionCompiler = 28
    // when target=27, 9.0 is ok
    versionTarget = 27
    // minSdkVersion 21
    // targetSdkVersion 30
    // if hope supporting 4.4
    // please modify it to 16
    minSdkVersion = 16
    versionCode = 101
    versionNameString = '2.3.6.20211117'
    javaSourceCompatibility = JavaVersion.VERSION_1_8
    javaTargetCompatibility = JavaVersion.VERSION_1_8
}
```