jiangdongguo/AndroidUSBCamera #402: How can I import libusbcamera library into my AndroidStudio project?

pedrofrau, 2022-05-08:
Hi, 

I've been struggling for several days trying to use both UVCCamera from saki4510t and AndroidUSBCamera from jiangdongguo. AndroidUSBCamera works as I download it from git, but I would like to make use of the library to use it on my app.

According to the readme file, we only need to add the jitpack repository, and the dependency of AndroidUSBCamera:2.3.7.

This is my build.gradle (:app) file for a test app I've been trying to compile:
`plugins {
    id 'com.android.application'
}

android {
    compileSdk 32

    defaultConfig {
        applicationId "com.example.camera_ext"
        minSdk 24
        targetSdk 32
        versionCode 1
        versionName "1.0"

        testInstrumentationRunner "androidx.test.runner.AndroidJUnitRunner"
    }

    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android-optimize.txt'), 'proguard-rules.pro'
        }
    }
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }
}

dependencies {
    implementation fileTree(include: ['*.jar', '*.aar'], dir: 'libs')
    implementation 'com.github.jiangdongguo:AndroidUSBCamera:2.3.7'
    implementation 'androidx.appcompat:appcompat:1.4.1'
    implementation 'com.google.android.material:material:1.5.0'
    implementation 'androidx.constraintlayout:constraintlayout:2.1.3'
    testImplementation 'junit:junit:4.13.2'
    androidTestImplementation 'androidx.test.ext:junit:1.1.3'
    androidTestImplementation 'androidx.test.espresso:espresso-core:3.4.0'
}

configurations {
    all*.exclude group: 'com.google.guava', module: 'listenablefuture'
}`

And this is my build.gradle(:Project):
`// Top-level build file where you can add configuration options common to all sub-projects/modules.
buildscript {
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:4.2.2'
        classpath 'com.github.dcendents:android-maven-gradle-plugin:2.1'
    }
}

plugins {
    id 'com.android.application' version '7.1.1' apply false
    id 'com.android.library' version '7.1.1' apply false
}

allprojects {
    repositories {
        //google()
        //jcenter()
        maven {
            url 'http://raw.github.com/saki4510t/libcommon/master/repository/'
            allowInsecureProtocol = true
        }
        maven { url 'https://jitpack.io' }

        flatDir {
            dirs 'libs', '../libusbcamera/libs'
        }
    }
}

task clean(type: Delete) {
    delete rootProject.buildDir
}`

My local.properties:
`sdk.dir=D\:\\Android
ndk.dir=D\:\\Android\\ndk\\android-ndk-r13b`

When I make the Gradle Sync, apparently it syncs correctly, but when I try to build the project (i.e. Make project) an error appears with no other description than `Missing required attribute: dependency groupId` 

I've seen in the issues that some people had the same problem, and the only answer I found is to put the following lines on the build.gradle file but still it does not work:
`configurations {
    all*.exclude group: 'com.google.guava', module: 'listenablefuture'
}`

I would like to know if there is a simple way to make use of libusbcamera library on my Android studio project. 

Thank you very much in advance

jiangdongguo, 2022-05-11:
please update to 2.3.8

ghelfer, 2022-06-15:
I updated to 2.3.8 and showed following error: Missing required attribute: dependency groupId

jiangdongguo, 2022-07-01:
Please update to new (3.0), and have a try.

closed 2022-07-01 by jiangdongguo

HoangBee102, 2023-10-27:
https://github.com/10dutel/YOLOv5_Android_USBCamera 
No any preview went open app, howto debug when OTG in use with usbcamera??? Pls help me