jiangdongguo/AndroidUSBCamera #426: 切换分辨率导致应用崩溃

ShshyDevooo, 2022-08-08:
以下是bugly捕捉信息
2022-08-08 10:56:36.377 3885-5096/com.jiangdg.ausbc E/CrashReport: #++++++++++Record By Bugly++++++++++#
2022-08-08 10:56:36.377 3885-5096/com.jiangdg.ausbc E/CrashReport: # You can use Bugly(http:\\bugly.qq.com) to get more Crash Detail!
2022-08-08 10:56:36.377 3885-5096/com.jiangdg.ausbc E/CrashReport: # PKG NAME: com.jiangdg.ausbc
2022-08-08 10:56:36.377 3885-5096/com.jiangdg.ausbc E/CrashReport: # APP VER: 3.2.4
2022-08-08 10:56:36.378 3885-5096/com.jiangdg.ausbc E/CrashReport: # SDK VER: 3.4.4-3.9.0
2022-08-08 10:56:36.378 3885-5096/com.jiangdg.ausbc E/CrashReport: # LAUNCH TIME: 2022-08-08 10:56:31
2022-08-08 10:56:36.378 3885-5096/com.jiangdg.ausbc E/CrashReport: # CRASH TYPE: NATIVE_CRASH
2022-08-08 10:56:36.378 3885-5096/com.jiangdg.ausbc E/CrashReport: # CRASH TIME: 2022-08-08 10:56:36
2022-08-08 10:56:36.379 3885-5096/com.jiangdg.ausbc E/CrashReport: # CRASH PROCESS: com.jiangdg.ausbc
2022-08-08 10:56:36.379 3885-5096/com.jiangdg.ausbc E/CrashReport: # CRASH THREAD: camera_manager(5340)
2022-08-08 10:56:36.379 3885-5096/com.jiangdg.ausbc E/CrashReport: # REPORT ID: 3eeca638-2634-46c6-abfb-049947ee7b1c
2022-08-08 10:56:36.379 3885-5096/com.jiangdg.ausbc E/CrashReport: # CRASH DEVICE: BAH3-AN10 UNROOT
2022-08-08 10:56:36.379 3885-5096/com.jiangdg.ausbc E/CrashReport: # RUNTIME AVAIL RAM:2832699392 ROM:93866409984 SD:93845438464
2022-08-08 10:56:36.380 3885-5096/com.jiangdg.ausbc E/CrashReport: # RUNTIME TOTAL RAM:5841080320 ROM:117790736384 SD:117769764864
2022-08-08 10:56:36.380 3885-5096/com.jiangdg.ausbc E/CrashReport: # EXCEPTION FIRED BY SI_QUEUE com.jiangdg.ausbc(3885)
2022-08-08 10:56:36.380 3885-5096/com.jiangdg.ausbc E/CrashReport: # CRASH STACK: 
2022-08-08 10:56:36.380 3885-5096/com.jiangdg.ausbc E/CrashReport: SIGABRT
    0x287400000f2d
    #00    pc 000000000003166c    /apex/com.android.runtime/lib64/bionic/libc.so (abort+160) [arm64-v8a::b91c775ccc9b0556e91bc575a2511cd0]
    #01    pc 00000000003a2d74    /apex/com.android.runtime/lib64/libart.so (_ZN3art7Runtime5AbortEPKc+2512) [arm64-v8a::762960c75ada06c48e438df6df29b0a1]
    #02    pc 0000000000003600    /system/lib64/libbase.so (_ZN7android4base10LogMessageD1Ev+684) [arm64-v8a::7e6f8e823512d07994e0f2250d7d708b]
    #03    pc 00000000003a2268    /apex/com.android.runtime/lib64/libart.so (_ZN3art7Runtime19DetachCurrentThreadEv+404) [arm64-v8a::762960c75ada06c48e438df6df29b0a1]
    #04    pc 000000000025be04    /apex/com.android.runtime/lib64/libart.so (_ZN3art2gc4Heap30PerformHomogeneousSpaceCompactEv+2516) [arm64-v8a::762960c75ada06c48e438df6df29b0a1]
    #05    pc 0000000000019624    /data/app/com.jiangdg.ausbc-DZju2LYnOmsSYwxF__ucaA==/lib/arm64/libUVCCamera.so (_ZN17UVCStatusCallback19uvc_status_callbackE16uvc_status_classii20uvc_status_attributePvmS2_+240) [arm64-v8a::00a99ee7a5025056bf99ffa603a4ac3c]
    #06    pc 000000000000c790    /data/app/com.jiangdg.ausbc-DZju2LYnOmsSYwxF__ucaA==/lib/arm64/libuvc.so (uvc_process_control_status+212) [arm64-v8a::821b57fa81492012beb0552253599eff]
    #07    pc 000000000000b070    /data/app/com.jiangdg.ausbc-DZju2LYnOmsSYwxF__ucaA==/lib/arm64/libuvc.so (_uvc_status_callback+120) [arm64-v8a::821b57fa81492012beb0552253599eff]
    #08    pc 000000000001019c    /data/app/com.jiangdg.ausbc-DZju2LYnOmsSYwxF__ucaA==/lib/arm64/libusb100.so (usbi_handle_transfer_completion+332) [arm64-v8a::7782f610e03566f5b9b5554e74ef42bc]
    #09    pc 00000000000153c0    /data/app/com.jiangdg.ausbc-DZju2LYnOmsSYwxF__ucaA==/lib/arm64/libusb100.so [arm64-v8a::7782f610e03566f5b9b5554e74ef42bc]
    #10    pc 0000000000010d64    /data/app/com.jiangdg.ausbc-DZju2LYnOmsSYwxF__ucaA==/lib/arm64/libusb100.so [arm64-v8a::7782f610e03566f5b9b5554e74ef42bc]
    #11    pc 00000000000107b4    /data/app/com.jiangdg.ausbc-DZju2LYnOmsSYwxF__ucaA==/lib/arm64/libusb100.so (libusb_handle_events_timeout_completed+396) [arm64-v8a::7782f610e03566f5b9b5554e74ef42bc]
    #12    pc 0000000000010f48    /data/app/com.jiangdg.ausbc-DZju2LYnOmsSYwxF__ucaA==/lib/arm64/libusb100.so (libusb_handle_events_completed+48) [arm64-v8a::7782f610e03566f5b9b5554e74ef42bc]
    #13    pc 00000000000115e0    /data/app/com.jiangdg.ausbc-DZju2LYnOmsSYwxF__ucaA==/lib/arm64/libusb100.so (libusb_control_transfer+364) [arm64-v8a::7782f610e03566f5b9b5554e74ef42bc]
    #14    pc 00000000000095a8    /data/app/com.jiangdg.ausbc-DZju2LYnOmsSYwxF__ucaA==/lib/arm64/libuvc.so (uvc_get_gain+64) [arm64-v8a::821b57fa81492012beb0552253599eff]
    #15    pc 0000000000013be8    /data/app/com.jiangdg.ausbc-DZju2LYnOmsSYwxF__ucaA==/lib/arm64/libUVCCamera.so (_ZN9UVCCamera15updateGainLimitERiS0_S0_+108) [arm64-v8a::00a99ee7a5025056bf99ffa603a4ac3c]
    #16    pc 0000000000023380    /data/app/com.jiangdg.ausbc-DZju2LYnOmsSYwxF__ucaA==/lib/arm64/libUVCCamera.so [arm64-v8a::00a99ee7a5025056bf99ffa603a4ac3c]
    #17    pc 0000000000012350    /apex/com.android.runtime/lib64/libart.so [arm64-v8a::762960c75ada06c48e438df6df29b0a1]
    #18    pc 0000000000009334    /apex/com.android.runtime/lib64/libart.so [arm64-v8a::762960c75ada06c48e438df6df29b0a1]
    #19    pc 00000000000181b4    /apex/com.android.runtime/lib64/libart.so (_ZN3art9ArtMethod6InvokeEPNS_6ThreadEPjjPNS_6JValueEPKc+252) [arm64-v8a::762960c75ada06c48e438df6df29b0a1]
    #20    pc 00000000001bf900    /apex/com.android.runtime/lib64/libart.so (_ZN3art11interpreter34ArtInterpreterToCompiledCodeBridgeEPNS_6ThreadEPNS_9ArtMethodEPNS_11ShadowFrameEtPNS_6JValueE+384) [arm64-v8a::762960c75ada06c48e438df6df29b0a1]
    #21    pc 00000000001babd0    /apex/com.android.runtime/lib64/libart.so (_ZN3art11interpreter6DoCallILb0ELb0EEEbPNS_9ArtMethodEPNS_6ThreadERNS_11ShadowFrameEPKNS_11InstructionEtPNS_6JValueE+912) [arm64-v8a::762960c75ada06c48e438df6df29b0a1]
    #22    pc 000000000048f910    /apex/com.android.runtime/lib64/libart.so (MterpInvokeDirect+400) [arm64-v8a::762960c75ada06c48e438df6df29b0a1]
    #23    pc 0000000000003914    /a
2022-08-08 10:56:36.380 3885-5096/com.jiangdg.ausbc E/CrashReport: #++++++++++++++++++++++++++++++++++++++++++#

jiangdongguo, 2022-08-09:
应该是使用摄像头底层崩了，是否必现？

ShshyDevooo, 2022-08-10:
> 

是必现的