jiangdongguo/AndroidUSBCamera #434: When Camera connect not display on Android Phone Sumsung A23

naeemiqbal696, 2022-08-21:


jiangdongguo, 2022-08-22:
please send your logs to me

sang981113, 2022-08-25:
I faced this problem too. It shows preview right when the app restart.

sang981113, 2022-08-25:
2022-08-25 17:10:09.674 2252-2252/com.jiangdg.ausbc I/AUSBC: Thread: main
2022-08-25 17:10:09.674 2252-2252/com.jiangdg.ausbc I/AUSBC: ++++++++Info->USBMonitor###UsbControlBlock#close:
2022-08-25 17:10:09.674 2252-2252/com.jiangdg.ausbc D/UsbDeviceConnectionJNI: close
2022-08-25 17:10:09.675 2252-2252/com.jiangdg.ausbc I/AUSBC: Thread: main
2022-08-25 17:10:09.677 2252-2252/com.jiangdg.ausbc I/AUSBC: ++++++++Info->CameraUvcStrategy###onDisconnect device = UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=46694,mProductId=61447,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=null,mProductName=HD WEBCAM,mVersion=0.14,mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@3881fdd, mHasAudioPlayback=false, mHasAudioCapture=false, mHasMidi=false, mHasVideoCapture=true, mHasVideoPlayback=true, mConfigurations=[
2022-08-25 17:10:09.677 2252-2252/com.jiangdg.ausbc I/AUSBC: UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=100,mInterfaces=[
2022-08-25 17:10:09.677 2252-2252/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=0,mAlternateSetting=0,mName=HD WEBCAM,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[
2022-08-25 17:10:09.677 2252-2252/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=129,mAttributes=3,mMaxPacketSize=16,mInterval=7]]
2022-08-25 17:10:09.677 2252-2252/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=0,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[]
2022-08-25 17:10:09.678 2252-2252/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=1,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
2022-08-25 17:10:09.678 2252-2252/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=130,mAttributes=5,mMaxPacketSize=5120,mInterval=1]]
2022-08-25 17:10:09.678 2252-2252/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=2,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
2022-08-25 17:10:09.678 2252-2252/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=130,mAttributes=5,mMaxPacketSize=3072,mInterval=1]]
2022-08-25 17:10:09.678 2252-2252/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=3,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
2022-08-25 17:10:09.678 2252-2252/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=130,mAttributes=5,mMaxPacketSize=1024,mInterval=1]]
2022-08-25 17:10:09.678 2252-2252/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=4,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
2022-08-25 17:10:09.678 2252-2252/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=130,mAttributes=5,mMaxPacketSize=512,mInterval=1]]]]
2022-08-25 17:10:09.678 2252-2252/com.jiangdg.ausbc V/AUSBC: Thread: main
2022-08-25 17:10:09.678 2252-2465/com.jiangdg.ausbc I/AUSBC: Thread: camera_manager
2022-08-25 17:10:09.679 2252-2252/com.jiangdg.ausbc V/AUSBC: ++++++++Info->USBMonitor###processDettach:
2022-08-25 17:10:09.679 2252-2465/com.jiangdg.ausbc I/AUSBC: ++++++++Info->CameraUvcStrategy###stopPreviewInternal
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: Thread: USBMonitor
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: ++++++++Info->CameraUvcStrategy###onDetach device = UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=46694,mProductId=61447,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=null,mProductName=HD WEBCAM,mVersion=0.14,mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@70c7bfe, mHasAudioPlayback=false, mHasAudioCapture=false, mHasMidi=false, mHasVideoCapture=true, mHasVideoPlayback=true, mConfigurations=[
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=100,mInterfaces=[
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=0,mAlternateSetting=0,mName=HD WEBCAM,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=129,mAttributes=3,mMaxPacketSize=16,mInterval=7]]
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=0,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[]
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=1,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=130,mAttributes=5,mMaxPacketSize=5120,mInterval=1]]
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=2,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=130,mAttributes=5,mMaxPacketSize=3072,mInterval=1]]
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=3,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=130,mAttributes=5,mMaxPacketSize=1024,mInterval=1]]
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=4,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
2022-08-25 17:10:09.680 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=130,mAttributes=5,mMaxPacketSize=512,mInterval=1]]]]
2022-08-25 17:10:09.683 2252-2465/com.jiangdg.ausbc I/AUSBC: Thread: camera_manager
2022-08-25 17:10:09.684 2252-2465/com.jiangdg.ausbc I/AUSBC: ++++++++Info->USBMonitor###UsbControlBlock#close:
[... cut ...]
2022-08-25 17:10:43.515 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=3,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
2022-08-25 17:10:43.515 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=130,mAttributes=5,mMaxPacketSize=1024,mInterval=1]]
2022-08-25 17:10:43.515 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=4,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
2022-08-25 17:10:43.515 2252-2454/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=130,mAttributes=5,mMaxPacketSize=512,mInterval=1]]]]

Here is my log until stuck.

jiangdongguo, 2022-08-26:
get permission failed in mUsbReceiver, please update to 3.2.7 and have a try

Luke-Tang, 2022-09-18:
问题在于USBMonitor的这行代码:
~~~
if (Build.VERSION.SDK_INT >= 31) {
					mPermissionIntent = PendingIntent.getBroadcast(context, 0, new Intent(ACTION_USB_PERMISSION), PendingIntent.FLAG_IMMUTABLE);
				} else {
					mPermissionIntent = PendingIntent.getBroadcast(context, 0, new Intent(ACTION_USB_PERMISSION), 0);
				}
~~~
FLAG_IMMUTABLE是指不让系统其他地方修改这个PendingIntent, 使用这个flag，会造成申请权限后，登记的receiver获取的intent中，获取不到携带的数据。

~~~
if (ACTION_USB_PERMISSION.equals(action)) {
				// when received the result of requesting USB permission
				synchronized (USBMonitor.this) {
					final UsbDevice device = intent.getParcelableExtra(UsbManager.EXTRA_DEVICE); //这里会一直返回null
					if (intent.getBooleanExtra(UsbManager.EXTRA_PERMISSION_GRANTED, false)) {  // 这里会一直得到false
						if (device != null) {
							// get permission, call onConnect
							if (DEBUG)
								XLogWrapper.w(TAG, "get permission success in mUsbReceiver");
							processConnect(device);
						}
					} else {
						// failed to get permission
						if (DEBUG)
							XLogWrapper.w(TAG, "get permission failed in mUsbReceiver");
						processCancel(device);
					}
				}
~~~

修改如下:

~~~

~~~
if (Build.VERSION.SDK_INT >= 31) {
					mPermissionIntent = PendingIntent.getBroadcast(context, 0, new Intent(ACTION_USB_PERMISSION), PendingIntent.FLAG_MUTABLE); 
                                   // 低版本的编译SDK会没有FLAG_MUTABLE， 替换为1<<25；高版本的编译SDK在android studio中也会提示错误，可以用注解@SuppressLint("WrongConstant")抑制错误提示或者直接使用1<<25
				} else {
					mPermissionIntent = PendingIntent.getBroadcast(context, 0, new Intent(ACTION_USB_PERMISSION), 0);
				}
~~~
~~~

jiangdongguo, 2022-09-29:
谢谢支持！