jiangdongguo/AndroidUSBCamera #438: 摄像头识别问题

LoveYouStill, 2022-08-24:
描述：

两个UVC摄像头(这里称为摄像头A和摄像头B)       两部测试机：VIVO IQQO U3X 和 小米8Lite

拉取项目 将需要的几个库集成到我的项目中 然后编译运行app到两部手机上 其中VIVO IQQO U3X是可以识别出这两个摄像头并正常显示画面的 而小米8Lite只可以识别摄像头A并显示画面 摄像头B一直识别不出来 一开始以为是摄像头的问题 就下载了一个USB双摄像头app测下 结果它是可以正常识别和显示的

后经过调试发现摄像头B打印的USBDevice的class和subClass都是0 导致摄像头逻辑判断不全被略过了(毕竟摄像头太多 可以理解) 所以后面就自己稍微处理了下让app能正常运行起来 但是并非最佳方式 若作者和大家有了更好的判断方式可以告知下 谢谢

[image]

[image]

[image]

CameraUtils文件修改处：

[image]

> 

> 

> 

fun isUsbCamera(device: UsbDevice?): Boolean {

        if(device != null) {
            return when {
                device.deviceClass == UsbConstants.USB_CLASS_VIDEO -> {
                    true
                }
                device.deviceClass == UsbConstants.USB_CLASS_MISC -> {
                    var isVideo = false
                    for (i in 0 until device.interfaceCount) {
                        val cls = device.getInterface(i).interfaceClass
                        if (cls == UsbConstants.USB_CLASS_VIDEO) {
                            isVideo = true
                            break
                        }
                    }
                    isVideo
                }
                device.manufacturerName?.contains("camera", true) == true -> {
                    true
                }
                else -> {
                    false
                }
            }
        }
        return false
    }

jiangdongguo, 2022-08-25:
看FQA第2点

If your uvc device can not be recognised, maybe you can try like this:

**First**, create a flie named `default_device_filter.xml` in app xml directory(If not exist, please create it.).

**then**, copying the fault value into it and adding your uvc device information.

```xml
<?xml version="1.0" encoding="utf-8"?>
<usb>
	<usb-device class="239" subclass="2" />	<!-- all device of UVC -->
	<!-- a few android 9.0 -->
	<usb-device class="14" subclass="9" />
	<usb-device class="2" subclass="0" />
	<usb-device class="6" subclass="-1" />
	<usb-device class="39" subclass="0" />
	<usb-device product-id="4836" vendor-id="9573" />
	<usb-device product-id="2229" vendor-id="1133" />
	<usb-device product-id="640" vendor-id="1409" />
	<usb-device product-id="258" vendor-id="9228" />
	<usb-device product-id="16981" vendor-id="19" />
    <!-- your uvc device -->
    <usb-device product-id="xxx" vendor-id="xxx" />
    <usb-device product-id="xxx" vendor-id="xxx" />
</usb>
```

LoveYouStill, 2022-08-26:
哦 好的 我只关注到前面一个判断 忽略了后一个判断 已经改正 十分感谢您的指出

jiangdongguo, 2022-08-26:
OK

closed 2022-08-26 by jiangdongguo