jiangdongguo/AndroidUSBCamera #444: How to exact frame ?

dFuong, 2022-08-26:
Can you help me about getting inputFrame ?
Please help me?

jiangdongguo, 2022-08-29:
inputFrame ?What does mean?

dFuong, 2022-08-29:
I want to process frame by Opencv so I need to exact frame.
I doesn't find out function getting frame.

jiangdongguo, 2022-08-29:
In CameraFragment or CameraActivity, you can use addPreviewDataCallBack to get NV21 bytes!

dFuong, 2022-08-29:
Thank you, I'll try it.
Best day for you.

jiangdongguo, 2022-08-29:
It's ok. Regards!

closed 2022-09-13 by jiangdongguo

ThuyKhanh, 2022-09-29:
Hello there,
I added addPreviewDataCallBack as below code but the callback is never called. Could you plz help ?
override fun onCameraConnected(camera: MultiCameraClient.Camera) {
        for ((position, cam) in mAdapter.data.withIndex()) {
            if (cam.getUsbDevice().deviceId == camera.getUsbDevice().deviceId) {
                val textureView = mAdapter.getViewByPosition(position, R.id.multi_camera_texture_view)
                cam.openCamera(textureView, getCameraRequest())
                cam.addPreviewDataCallBack(previewDataCallBack)
                cam.setCameraStateCallBack(this)
                break
            }
        }