jiangdongguo/AndroidUSBCamera #446: 运行Demo程序，切换USB摄像头没有反应

EthanCo, 2022-08-29:
Android平板接着两个同一型号的USB摄像头 (海康威视 HIK 1080P Camera )
运行Demo程序，点击切换摄像头，没有反应。

EthanCo, 2022-08-29:
我发现这两个摄像头读取到的`dev.productName`都是`HIK 1080P Camera`，而读取到的`dev.deviceName`和`dev.deviceId`是不同的。
然后我对`DemoFragment.kt`中的`showUsbDevicesDialog()`进行了修改  

原本的代码
```kotlin
val devName =
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP && !dev.productName.isNullOrEmpty()) {
    dev.productName
} else {
    "${dev.deviceName}(${dev.deviceId})"
}
val curDevName =
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP && !curDevice?.productName.isNullOrEmpty()) {
    curDevice!!.productName
} else {
    "${curDevice?.deviceName}(${curDevice?.deviceId})"
}
```

修改为
```kotlin
val devName =
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP && !dev.productName.isNullOrEmpty()) {
    "${dev.deviceName}(${dev.deviceId})"//dev.productName
} else {
    "${dev.deviceName}(${dev.deviceId})"
}
val curDevName =
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP && !curDevice?.productName.isNullOrEmpty()) {
    "${curDevice?.deviceName}(${curDevice?.deviceId})"//curDevice!!.productName
} else {
    "${curDevice?.deviceName}(${curDevice?.deviceId})"
}
```

重新运行程序后，再切换摄像头，就可以正常切换了

jiangdongguo, 2022-08-29:
谢谢。下个版本体现

closed 2022-08-30 by jiangdongguo