jiangdongguo/AndroidUSBCamera #449: MultiCameraActivity中开启多摄像头 分辨率设置无效

liu408309046, 2022-08-31:
在继承MultiCameraActivity的 onCameraConnected 中开启双摄像头，摄像头正常开启预览但是分别率设置无效
camera.openCamera(secondCameraTextureView, getCameraRequest(2048,1536));

    private CameraRequest getCameraRequest(int width,int height) {
        return new CameraRequest.Builder().setPreviewWidth(width).setPreviewHeight(height).create();
    }

//   camera.updateResolution(2048,1536); 使用更新分辨率报错
 
    java.lang.UnsupportedOperationException: open failed:result=-99----->id_camera=2297824768;venderId=3034;productId=12342;fileDescriptor=160;busNum=4;devAddr=5;usbfs=/dev/bus/usb
    Exception：调用nativeConnect返回值：-99
        at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:215)

liu408309046, 2022-09-01:
> 是哪个版本？3.2.8？

是的3.2.8

liu408309046, 2022-09-02:
@jiangdongguo  3.2.6设置分辨率是有效的  3.2.8设置无效

jiangdongguo, 2022-09-05:
拉下最新代码试试

closed 2022-09-06 by liu408309046