jiangdongguo/AndroidUSBCamera #455: can this library be used with jetpack compose?

shanerodrigues, 2022-09-08:
Hello, I'm new to android development. I have a jetpack compose project and want to use AUSBC to use my external camera in the app to take photos. Is it possible to use AUSBC with jetpack compose? If so, can you provide some info on how to do it?

jiangdongguo, 2022-09-09:
Maybe, it can, but I haven't learn about jetpack compose.

shanerodrigues, 2022-09-11:
Ok, thanks for your help.

closed 2022-09-13 by jiangdongguo

Luke-Tang, 2022-09-14:
Yes, it can. I am using it in compose.

shanerodrigues, 2022-09-14:
@Luke-Tang, thanks for letting me know! I would really appreciate it if you could provide an example of how you did it.

Luke-Tang, 2022-09-17:
The code is roughly as follows,  remember grant camera permission

class MainActivity : ComponentActivity() {

    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)

        setContent {
            YourAppTheme(darkTheme = true) {
                // A surface container using the 'background' color from the theme
                Surface(
                    modifier = Modifier.fillMaxSize(),
                    color = MaterialTheme.colorScheme.background
                ) {
                    YourApp()
                }
            }
        }
    }
}

@Composable
fun rememberCameraClient(context: Context): CameraClient = remember {
    CameraClient.newBuilder(context).apply {
        setEnableGLES(true)
        setRawImage(false)
        setCameraStrategy(CameraUvcStrategy(context))
        setCameraRequest(
            CameraRequest.Builder()
                .setFrontCamera(true)
                .setPreviewWidth(960)
                .setPreviewHeight(720)
                .create()
        )
        openDebug(true)
    }.build()
}

@Composable
fun YourApp() {
    val context = LocalContext.current
    Box(modifier = Modifier.fillMaxSize()) {
        UVCCameraPreview(rememberCameraClient(context))
    }
}

@Composable
fun UVCCameraPreview(cameraClient : CameraClient) {
    AndroidView(
        factory = { ctx ->
            AspectRatioSurfaceView(ctx).apply {
                this.holder.addCallback(object : Callback {
                    override fun surfaceCreated(holder: SurfaceHolder) {
                        cameraClient.openCamera(this@apply)
                    }

                    override fun surfaceChanged(
                        holder: SurfaceHolder,
                        format: Int,
                        width: Int,
                        height: Int
                    ) {
                        cameraClient.setRenderSize(width, height)
                    }

                    override fun surfaceDestroyed(holder: SurfaceHolder) {
                        cameraClient.closeCamera()
                    }
                })
            }
        }
    ) {
//        AspectRatioTextureView
    }
}

shanerodrigues, 2022-09-19:
@Luke-Tang thank you very much for the code. I've implemented it in this repo here - https://github.com/shanerodrigues/compose-uvc-camera. 

However, I get errors when running and the app crashes.

2022-09-19 12:20:42.900 14461-14561/? E/AndroidRuntime: FATAL EXCEPTION: gl_render
    Process: com.example.camerauvctest, PID: 14461
    java.lang.IllegalArgumentException: Make sure the SurfaceView or associated SurfaceHolder has a valid Surface
        at android.opengl.EGL14._eglCreateWindowSurface(Native Method)
        at android.opengl.EGL14.eglCreateWindowSurface(EGL14.java:263)
        at com.jiangdg.ausbc.render.env.EGLEvn.setupSurface(EGLEvn.kt:101)
        at com.jiangdg.ausbc.render.internal.ScreenRender.setupSurface(ScreenRender.kt:36)
        at com.jiangdg.ausbc.render.RenderManager.handleMessage(RenderManager.kt:122)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)

2022-09-19 12:20:43.052 14461-14556/? E/libusb/usbfs: [14556*android_usbfs.c:2881:handle_control_completion]:LIBUSB_TRANSFER_STALL
2022-09-19 12:20:43.053 14461-14560/? E/libusb/usbfs: [14560*android_usbfs.c:1776:release_interface]:release interface failed, error -1 errno 22
2022-09-19 12:20:43.054 14461-14560/? E/UVCCamera: [14560*UVCCamera.cpp:172:connect]:could not open camera:err=-99
2022-09-19 12:20:43.058 14461-14560/? E/CameraUvc:  preview failed, err = open failed:result=-99----->id_camera=2802226816;venderId=11030;productId=26249;fileDescriptor=84;busNum=1;devAddr=58;usbfs=/dev/bus/usb
    Exception：调用nativeConnect返回值：-99
    java.lang.UnsupportedOperationException: open failed:result=-99----->id_camera=2802226816;venderId=11030;productId=26249;fileDescriptor=84;busNum=1;devAddr=58;usbfs=/dev/bus/usb
    Exception：调用nativeConnect返回值：-99
        at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:215)
        at com.jiangdg.ausbc.camera.CameraUvcStrategy.createCamera(CameraUvcStrategy.kt:132)
        at com.jiangdg.ausbc.camera.CameraUvcStrategy.startPreviewInternal(CameraUvcStrategy.kt:113)
        at com.jiangdg.ausbc.camera.ICameraStrategy.handleMessage(ICameraStrategy.kt:96)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2022-09-19 12:20:43.095 12096-12096/? E/jindi: 监听：false---type:0
2022-09-19 12:20:43.098 12096-12096/? E/jindi: 隐藏
2022-09-19 12:20:43.103 12096-12096/? E/jindi: taskcom.example.camerauvctest
2022-09-19 12:20:43.105 12096-12096/? E/jindi: 监听：true---type:0
2022-09-19 12:20:43.110 12096-12096/? E/jindi: 显示
2022-09-19 12:20:43.194 12096-12096/? E/jindi: taskcom.example.camerauvctest
2022-09-19 12:20:43.213 12096-12096/? E/jindi: taskcom.example.camerauvctest

Do you have an idea as to why this is happening?

shanerodrigues, 2022-09-19:
@Luke-Tang, thank you very much for your help. I've got it working perfectly! in case anyone wants a working compose version, here's the link to the repo https://github.com/shanerodrigues/compose-uvc-camera

shanerodrigues, 2022-09-19:
@Luke-Tang one last question, if you don't mind. I'm not sure how to implement captureImage. I've got this code, any tips on how to do it?
```
fun YourApp() {

    var cClient = rememberCameraClient(LocalContext.current)

    val context = LocalContext.current
    Box(modifier = Modifier.fillMaxSize()) {
        UVCCameraPreview(rememberCameraClient(context))
        Column(
            modifier = Modifier.fillMaxSize(),
            verticalArrangement = Arrangement.Bottom,
            horizontalAlignment = Alignment.CenterHorizontally,
        ){
            Button(onClick={clickPhoto(cClient)}){
                Text("Take Picture")
            }
        }

    }

}

fun clickPhoto(cameraClient : CameraClient){
    Log.i("check", "inside click photo")
    ToastUtils.show("inside click photo")
    cameraClient?.captureImage(object : ICaptureCallBack {

        override fun onBegin() {
            ToastUtils.show("OnBegin")
            Log.i(TAG, "onBegin")
        }

        override fun onError(error: String?) {
            ToastUtils.show(error ?: "未知异常")
            Log.i(TAG, "onError")
        }

        override fun onComplete(path: String?) {
            ToastUtils.show("OnComplete")
            Log.i(TAG, "onComplete")
        }
    })
}
```

Luke-Tang, 2022-09-22:
@shanerodrigues 
Yes, that's correct.   You may be pay attention to storage permision， especially  target version greater than android10. The build target of this library is 27。

shanerodrigues, 2022-09-28:
@Luke-Tang would you know how to take a high-resolution photo? In rememberCameraClient I've set the resolution to 2592x1944 which my camera supports, but the photos are taken with the preview width of 800x600. Does it have something to do with the AspectRatioSurfaceView?

sheenhill, 2024-01-29:
compose demo. From 3.2.9

``` kotlin

class Test6Activity : ComponentActivity() {

    lateinit var client: MultiCameraClient

    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        client = MultiCameraClient(this, object : IDeviceConnectCallBack {
            override fun onAttachDev(device: UsbDevice?) {
                LogUtil.d("onAttachDev >>> $device")
                if (device != null) {
                    LogUtil.d("device.id >>> ${device.productId}")
                }
                device ?: return
                if (mCameraMap.containsKey(device.productId)) {
                    return
                }
                CameraUVC(this@Test6Activity, device).apply {
                    mCameraMap[device.productId] = this
                    client.requestPermission(device)
                }
            }

            override fun onDetachDec(device: UsbDevice?) {
                LogUtil.d("onDetachDec >>> $device")
            }

            override fun onConnectDev(device: UsbDevice?, ctrlBlock: USBMonitor.UsbControlBlock?) {
                LogUtil.d("onConnectDev >>> $device")
                mCameraMap[device?.productId]?.apply {
                    setUsbControlBlock(ctrlBlock)
                }
            }

            override fun onDisConnectDec(
                device: UsbDevice?, ctrlBlock: USBMonitor.UsbControlBlock?
            ) {
                LogUtil.d("onDisConnectDec >>> $device")
            }

            override fun onCancelDev(device: UsbDevice?) {
                LogUtil.d("onCancelDev >>> $device")
            }

        })
        client.register()
        setContent {
            CloudCanteenPlusTheme {
                Surface(
                    modifier = Modifier.fillMaxSize(), color = MaterialTheme.colors.background
                ) {
                    MyApp()
                }
            }
        }
    }

    private val mCameraMap = hashMapOf<Int, MultiCameraClient.ICamera>()

    @Composable
    private fun MyApp() {
        Box(modifier = Modifier.fillMaxWidth()) {
            var device1 by remember {
                mutableStateOf<MultiCameraClient.ICamera?>(null)
            }
            LaunchedEffect(key1 = true, block = {
                delay(1000)
                device1 = mCameraMap[1002] // 1002 自己的摄像头
            })
            LogUtil.d("MyApp  device1 >>> $device1")
            device1?.let {
                UVCCameraPreview(
                    modifier = Modifier.fillMaxSize(),
                    it
                )
            }

        }
    }

    @Composable
    private fun UVCCameraPreview(
        modifier: Modifier = Modifier,
        cameraClient: MultiCameraClient.ICamera
    ) {
        Box(
            modifier = modifier
                .aspectRatio(4f / 3f)
        ) {
            AndroidView(factory = { ctx ->
                AspectRatioSurfaceView(ctx).apply {
                    this.holder.addCallback(object : SurfaceHolder.Callback {
                        override fun surfaceCreated(holder: SurfaceHolder) {
                            val request = CameraRequest.Builder()
                                .setPreviewWidth(640)
                                .setPreviewHeight(480)
                                .create()
                            cameraClient.openCamera(this@apply, request)
                            cameraClient.addPreviewDataCallBack(object : IPreviewDataCallBack {
                                override fun onPreviewData(
                                    data: ByteArray?,
                                    width: Int,
                                    height: Int,
                                    format: IPreviewDataCallBack.DataFormat
                                ) {
                                    //  data:4915207  width:1280  height:960  format:RGBA
                                    LogUtil.d("onPreviewData >>> data:${data?.size}  width:$width  height:$height  format:$format")
                                    // Receive Data
                                }
                            })
                        }

                        override fun surfaceChanged(
                            holder: SurfaceHolder, format: Int, width: Int, height: Int
                        ) {
                            cameraClient.setRenderSize(width, height)
                        }

                        override fun surfaceDestroyed(holder: SurfaceHolder) {
                            cameraClient.closeCamera()
                        }
                    })
                }
            }) {}
        }
    }
}
```