jiangdongguo/AndroidUSBCamera #458: Unable to build

Petri-Leinonen, 2022-09-14:
I'm trying to build this library using the gradlew.bat, but I'm getting the following error:

```
C:\Users\cnpet\Documents\73Health\AndroidUSBCamera-master>gradlew.bat

> Configure project :libuvc
executing ndkBuild
WARNING:: Using flatDir should be avoided because it doesn't support any meta-data formats.

FAILURE: Build failed with an exception.

* Where:
Build file 'C:\Users\cnpet\Documents\73Health\AndroidUSBCamera-master\libuvc\build.gradle' line: 55

* What went wrong:
A problem occurred evaluating project ':libuvc'.
> C:\Users\cnpet\Documents\73Health\AndroidUSBCamera-master\local.properties (Mõõritettyõ tiedostoa ei l÷ydy)

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
```

I'm building this because I need to get new `app-release.apk` file, with the added device in `default_device_filter.xml`. My camera didn't work with the default list of devices, so I need to add my camera information.

jiangdongguo, 2022-09-15:
Please check your local.properties file and add `ndk.dir` path

Petri-Leinonen, 2022-09-15:
> Please check your local.properties file and add `ndk.dir` path

I opened the project using Android Studio and it created the `local.properties` file automatically with correct paths. This seems to be the easiest fix for this.

closed 2022-09-15 by Petri-Leinonen

Zeus178, 2022-09-26:
> Please check your local.properties file and add `ndk.dir` path

Please, i don't know what the ndk path

Zeus178, 2022-09-26:
> > Please check your local.properties file and add `ndk.dir` path
> 
> I opened the project using Android Studio and it created the `local.properties` file automatically with correct paths. This seems to be the easiest fix for this.

> Please, i don't know what the ndk path

Please, i don't know what the ndk path

Petri-Leinonen, 2022-09-26:
> 

Create a file `local.properties` to the root of project. Then add the following lines to the file (check that the path names are correct for you - it might be in different directory):

sdk.dir=C:\\Users\\YOUR_USERNAME\\AppData\\Local\\Android\\Sdk
ndk.dir=C:\\Users\\YOUR_USERNAME\\AppData\\Local\\Android\\Sdk\\ndk\\21.4.7075529

Zeus178, 2022-09-26:
> eck that the path names are correct for you - it might be in different dir

Thanks, I have followed and  resolve my issue