jiangdongguo/AndroidUSBCamera #459: Not working in Oppo and Huawei

KanwalKhan-20, 2022-09-16:
Hi 
The latest APK is not working in the following mobiles.
Huawei Model Nova 3i - Android Version 9   
Oppo Model f9 android 10

jiangdongguo, 2022-09-16:
Pleas send me your logs at location Android/data/com.jiangdg.ausbc.

KanwalKhan-20, 2022-09-20:
> Pleas send me your logs at location `Android/data/com.jiangdg.ausbc`.

[JJCamera-2022-09-15.log zip](https://github.com/jiangdongguo/AndroidUSBCamera/files/9604129/JJCamera-2022-09-15.log.zip)

Oppo F9

[huawei  y6p    JJCamera-2022-09-19.log](https://github.com/jiangdongguo/AndroidUSBCamera/files/9604113/huawei.y6p.JJCamera-2022-09-19.log)

KanwalKhan-20, 2022-09-22:
Hi @jiangdongguo  
any luck regarding the issue ?

jiangdongguo, 2022-09-22:
Please update to 3.2.9, and have a try

KanwalKhan-20, 2022-09-22:
Hi @jiangdongguo 
[image]