jiangdongguo/AndroidUSBCamera #460: Make sure the SurfaceView or associated SurfaceHolder has a valid Surface

kingzcheung, 2022-09-22:
请问这个问题是什么原因？

我没有看到有指定设备id的相关代码，是不是如果有多设备必须使用多路的调用方式？

```
 E  FATAL EXCEPTION: gl_render
                                                                                                    Process: com.example.triscdemo, PID: 25890
                                                                                                    java.lang.IllegalArgumentException: Make sure the SurfaceView or associated SurfaceHolder has a valid Surface
                                                                                                    	at android.opengl.EGL14._eglCreateWindowSurface(Native Method)
                                                                                                    	at android.opengl.EGL14.eglCreateWindowSurface(EGL14.java:263)
                                                                                                    	at com.jiangdg.ausbc.render.env.EGLEvn.setupSurface(EGLEvn.kt:101)
                                                                                                    	at com.jiangdg.ausbc.render.internal.ScreenRender.setupSurface(ScreenRender.kt:36)
                                                                                                    	at com.jiangdg.ausbc.render.RenderManager.handleMessage(RenderManager.kt:122)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                                                    	at android.os.Looper.loop(Looper.java:193)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:65)
2022-09-22 18:33:55.414 25890-26056 EGLEvn                  com.example.triscdemo                I  Init EGL Success!
2022-09-22 18:33:55.420 25890-26053 CameraUvc               com.example.triscdemo                I   start preview success!!!, id(null)null
2022-09-22 18:33:55.423 25890-26056 EGLEvn                  com.example.triscdemo                I  setupSurface Success!
2022-09-22 18:33:55.433 25890-26055 CameraUvc               com.example.triscdemo                I   start preview success!!!, id(null)null
2022-09-22 18:33:55.461 25890-26056 AbstractRender          com.example.triscdemo                I  init surface texture render success!
2022-09-22 18:33:55.471 25890-26054 Process                 com.example.triscdemo                I  Sending signal. PID: 25890 SIG: 9
2022-09-22 18:33:55.855  1389-1640  InputDispatcher         system_process                       E  channel 'f6a576d com.example.triscdemo/com.example.triscdemo.MainActivity (server)' ~ Channel is unrecoverably broken and will be disposed!
```

我没有看到有指定设备id的相关代码，是不是如果有多设备必须使用多路的调用方式？

jiangdongguo, 2022-09-23:
多路使用Multi开头的类

kingzcheung, 2022-09-23:
上面这个问题，应该是初始化的一些问题导致的，切换compose路由就必出现。

我使用的是jetpack compose。单路我看issue有人做了。

kingzcheung, 2022-09-23:
> 多路使用Multi开头的类

这个例子，在多路由页面下就会复现。

https://github.com/shanerodrigues/compose-uvc-camera

closed 2024-09-10 by kingzcheung