jiangdongguo/AndroidUSBCamera #464: 分辨率切换的时候，有概率出现失败并且卡住

liujie136918, 2022-09-26:
在定制开发版上预览正常，切换分辨率的时候偶尔会失败，失败后会卡死

2022-09-26 17:41:07.249 2804-2804/com.jiangdg.ausbc I/AUSBC: Thread: main
2022-09-26 17:41:07.249 2804-2804/com.jiangdg.ausbc I/AUSBC: ++++++++Info->RenderManager###camera render frame rate is 26 fps
2022-09-26 17:41:07.258 2804-3343/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-0](this:0x9720e000,id:0,api:1,p:2804,c:2804) queueBuffer: fps=25.11 dur=1035.58 max=44.36 min=34.77
2022-09-26 17:41:07.408 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: fps=24.99 dur=1000.60 max=44.20 min=36.01
2022-09-26 17:41:08.285 2804-2804/com.jiangdg.ausbc I/AUSBC: Thread: main
2022-09-26 17:41:08.285 2804-2804/com.jiangdg.ausbc I/AUSBC: ++++++++Info->RenderManager###camera render frame rate is 26 fps
2022-09-26 17:41:08.294 2804-3343/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-0](this:0x9720e000,id:0,api:1,p:2804,c:2804) queueBuffer: fps=25.09 dur=1036.45 max=43.53 min=35.16
2022-09-26 17:41:08.444 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: fps=25.10 dur=1035.78 max=42.30 min=34.76
2022-09-26 17:41:09.321 2804-2804/com.jiangdg.ausbc I/AUSBC: Thread: main
2022-09-26 17:41:09.321 2804-2804/com.jiangdg.ausbc I/AUSBC: ++++++++Info->RenderManager###camera render frame rate is 26 fps
2022-09-26 17:41:09.330 2804-3343/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-0](this:0x9720e000,id:0,api:1,p:2804,c:2804) queueBuffer: fps=25.09 dur=1036.14 max=42.15 min=37.15
2022-09-26 17:41:09.487 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: slot 1 is dropped, handle=0x9756a180
2022-09-26 17:41:09.488 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: fps=24.91 dur=1043.61 max=47.66 min=34.89
2022-09-26 17:41:09.521 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: slot 2 is dropped, handle=0xa792b880
2022-09-26 17:41:09.560 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: slot 1 is dropped, handle=0x9756a180
2022-09-26 17:41:09.587 2804-2804/com.jiangdg.ausbc D/ViewRootImpl[MainActivity]: hardware acceleration = true , fakeHwAccelerated = false, sRendererDisabled = false, forceHwAccelerated = false, sSystemRendererDisabled = false
2022-09-26 17:41:09.609 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: slot 2 is dropped, handle=0xa792b880
2022-09-26 17:41:09.648 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: slot 1 is dropped, handle=0x9756a180
2022-09-26 17:41:09.674 2804-2815/com.jiangdg.ausbc I/m.jiangdg.ausb: Background concurrent copying GC freed 1198(144KB) AllocSpace objects, 4(1808KB) LOS objects, 23% free, 6MB/8MB, paused 174us total 110.299ms
2022-09-26 17:41:09.682 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: slot 2 is dropped, handle=0xa792b880
2022-09-26 17:41:09.726 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: slot 1 is dropped, handle=0x9756a180
2022-09-26 17:41:09.765 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: slot 2 is dropped, handle=0xa792b880
2022-09-26 17:41:09.796 2804-2847/com.jiangdg.ausbc D/Surface: Surface::connect(this=0x9720f800,api=1)
2022-09-26 17:41:09.796 2804-2804/com.jiangdg.ausbc D/Surface: Surface::allocateBuffers(this=0x9720f800)
2022-09-26 17:41:09.801 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: slot 1 is dropped, handle=0x9756a180
2022-09-26 17:41:09.940 2804-3343/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-0](this:0x9720e000,id:0,api:1,p:2804,c:2804) queueBuffer: slot 2 is dropped, handle=0x9756a9c0
2022-09-26 17:41:10.360 2804-2804/com.jiangdg.ausbc I/AUSBC: Thread: main
2022-09-26 17:41:10.360 2804-2804/com.jiangdg.ausbc I/AUSBC: ++++++++Info->RenderManager###camera render frame rate is 17 fps
2022-09-26 17:41:10.368 2804-3343/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-0](this:0x9720e000,id:0,api:1,p:2804,c:2804) queueBuffer: fps=16.38 dur=1037.54 max=423.28 min=18.62
2022-09-26 17:41:10.517 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: fps=25.25 dur=1029.60 max=63.06 min=27.61
2022-09-26 17:41:11.397 2804-2804/com.jiangdg.ausbc I/AUSBC: Thread: main
2022-09-26 17:41:11.397 2804-2804/com.jiangdg.ausbc I/AUSBC: ++++++++Info->RenderManager###camera render frame rate is 26 fps
2022-09-26 17:41:11.407 2804-3343/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-0](this:0x9720e000,id:0,api:1,p:2804,c:2804) queueBuffer: fps=25.02 dur=1039.16 max=47.44 min=32.73
2022-09-26 17:41:11.560 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: fps=24.93 dur=1042.72 max=46.74 min=33.00
2022-09-26 17:41:11.756 2804-2815/com.jiangdg.ausbc I/m.jiangdg.ausb: Background concurrent copying GC freed 416(159KB) AllocSpace objects, 5(2MB) LOS objects, 22% free, 6MB/8MB, paused 493us total 140.607ms
2022-09-26 17:41:11.782 2804-3343/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-0](this:0x9720e000,id:0,api:1,p:2804,c:2804) queueBuffer: slot 2 is dropped, handle=0x9756a9c0
2022-09-26 17:41:12.433 2804-2804/com.jiangdg.ausbc I/AUSBC: Thread: main
2022-09-26 17:41:12.433 2804-2804/com.jiangdg.ausbc I/AUSBC: ++++++++Info->RenderManager###camera render frame rate is 26 fps
2022-09-26 17:41:12.440 2804-3343/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-0](this:0x9720e000,id:0,api:1,p:2804,c:2804) queueBuffer: fps=25.16 dur=1033.42 max=58.81 min=17.99
2022-09-26 17:41:12.596 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: fps=25.10 dur=1035.84 max=47.13 min=29.17
2022-09-26 17:41:13.467 2804-2804/com.jiangdg.ausbc I/AUSBC: Thread: main
2022-09-26 17:41:13.467 2804-2804/com.jiangdg.ausbc I/AUSBC: ++++++++Info->RenderManager###camera render frame rate is 26 fps
2022-09-26 17:41:13.476 2804-3343/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-0](this:0x9720e000,id:0,api:1,p:2804,c:2804) queueBuffer: fps=25.11 dur=1035.29 max=46.21 min=31.28
2022-09-26 17:41:13.631 2804-3345/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-11](this:0x9725a000,id:11,api:2,p:2804,c:2804) queueBuffer: fps=25.11 dur=1035.45 max=46.32 min=32.98
2022-09-26 17:41:13.757 2804-2804/com.jiangdg.ausbc I/AUSBC: Thread: main
2022-09-26 17:41:13.758 2804-2804/com.jiangdg.ausbc I/AUSBC: ++++++++Info->CameraClient###updateResolution size = 1280x720
2022-09-26 17:41:13.769 2804-2804/com.jiangdg.ausbc I/AUSBC: Thread: main
2022-09-26 17:41:13.769 2804-2804/com.jiangdg.ausbc I/AUSBC: ++++++++Info->CameraClient###closeCamera...
2022-09-26 17:41:13.773 2804-3342/com.jiangdg.ausbc W/OrientationEventListener: Cannot detect sensors. Invalid disable
2022-09-26 17:41:13.777 2804-3343/com.jiangdg.ausbc I/AUSBC: Thread: gl_render
2022-09-26 17:41:13.777 2804-3343/com.jiangdg.ausbc I/AUSBC: ++++++++Info->AbstractRender###release surface texture render success!
2022-09-26 17:41:13.783 2804-3342/com.jiangdg.ausbc I/AUSBC: Thread: camera_manager
2022-09-26 17:41:13.783 2804-3342/com.jiangdg.ausbc I/AUSBC: ++++++++Info->CameraUvc###stopPreviewInternal
2022-09-26 17:41:13.795 2804-3343/com.jiangdg.ausbc D/Surface: Surface::disconnect(this=0x9707c000,api=1)
2022-09-26 17:41:13.796 2804-3343/com.jiangdg.ausbc I/BufferQueueProducer: [SurfaceTexture-0-2804-0](this:0x9720e000,id:0,api:1,p:2804,c:2804) disconnect(P): api 1
2022-09-26 17:41:13.796 2804-2847/com.jiangdg.ausbc D/Surface: Surface::disconnect(this=0x9720f800,api=1)
[... cut ...]
2022-09-26 17:41:38.860 2804-3387/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=5056,mInterval=1]]
2022-09-26 17:41:38.860 2804-3387/com.jiangdg.ausbc I/AUSBC: UsbInterface[mId=1,mAlternateSetting=11,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
2022-09-26 17:41:38.860 2804-3387/com.jiangdg.ausbc I/AUSBC: UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=5116,mInterval=1]]]]
2022-09-26 17:41:38.861 2804-3387/com.jiangdg.ausbc V/AUSBC: Thread: camera_manager
2022-09-26 17:41:38.861 2804-3387/com.jiangdg.ausbc V/AUSBC: ++++++++Info->UVCCamera###close:finished

jiangdongguo, 2022-09-27:
哪个版本？

liujie136918, 2022-09-28:
最新的版本 3.2.8

QGB, 2022-10-22:
@jiangdongguo  我一打开app 就卡住

EthanCo, 2022-11-10:
我也有遇到这个问题