jiangdongguo/AndroidUSBCamera #468: android 11 is not working

Luke-Tang, 2022-09-28:
使用测试的AUSBC.apk，版本为3.2.7。预览一直黑屏。 但可以获取到可选的预览尺寸。
手机型号为: Redmi Note 10， android 11.

[AUSBC_v3.2.7_2022-09-28.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/9661019/AUSBC_v3.2.7_2022-09-28.txt)

Luke-Tang, 2022-09-28:
同一个手机， 拉最新的代码编译，同样现象。下面是最新的log
[AUSBC_v3.3.0_2022-09-28.log](https://github.com/jiangdongguo/AndroidUSBCamera/files/9661417/AUSBC_v3.3.0_2022-09-28.log)

annaijinwind, 2022-09-29:
https://github.com/jiangdongguo/AndroidUSBCamera/issues/434#issuecomment-1250231199

Luke-Tang, 2022-09-29:
@annaijinwind 
不是同一个问题。 #434是权限问题。 这个是有权限，并且log中显示预览成功了，但实际没画面

jiangdongguo, 2022-09-29:
多路能不能正常预览？就是左上角菜单里的功能

Luke-Tang, 2022-09-30:
不能预览。 但有时候进多路的时候会弹窗提示，-99, 但不是每次都提示。附件是log
[AUSBC_v3.3.0_2022-09-30.log](https://github.com/jiangdongguo/AndroidUSBCamera/files/9680171/AUSBC_v3.3.0_2022-09-30.log)

jiangdongguo, 2022-09-30:
Redmi Note 10有点拉跨，你换个其他品牌的android11试试

Luke-Tang, 2022-09-30:
我得找找看。 目前鸿蒙2.0系统的几款机器都正常使用

Luke-Tang, 2022-09-30:
荣耀畅玩30 也是同样的现象