jiangdongguo/AndroidUSBCamera #471: How to set the render width and height of the preview?

shanerodrigues, 2022-09-29:
My usb camera supports taking 2592x1944 resolution photos, but when I load the preview, the preview renders in 800x600 and takes 800x600 photos. Is there a way to set the render size so I can take 2592x1944 resolution photos? I'm using jetpack compose.
I've got a log here - https://pastebin.com/H9aPQt91
I set the preview width and height in cameraClient as 2592x1944.
```kotlin
@Composable
fun rememberCameraClient(context: Context): CameraClient = remember {
    CameraClient.newBuilder(context).apply {
        setEnableGLES(true)
        setRawImage(false)
        setCameraStrategy(CameraUvcStrategy(context))
        setCameraRequest(
            CameraRequest.Builder()
                .setFrontCamera(false)
                .setPreviewWidth(2592)
                .setPreviewHeight(1944)
                .create()
        )
        openDebug(true)
    }.build()
}
@Composable
fun UVCCameraPreview(cameraClient : CameraClient) {
    val camView: IAspectRatio? = null
    camView?.setAspectRatio(2592, 1944)
    AndroidView(
        modifier = Modifier.size(500.dp),
        factory = { ctx ->
            AspectRatioSurfaceView(ctx).apply {
                this.holder.addCallback(object : SurfaceHolder.Callback {
                    override fun surfaceCreated(holder: SurfaceHolder) {

                        Log.i("testing", "camView is ${camView}")
                        cameraClient.openCamera(this@apply)
                    }

                    override fun surfaceChanged(
                        holder: SurfaceHolder,
                        format: Int,
                        width: Int,
                        height: Int
                    ) {
                        cameraClient.setRenderSize(2592, 1944)
                    }

                    override fun surfaceDestroyed(holder: SurfaceHolder) {
                        cameraClient.closeCamera()
                    }
                })
            }
        }
    ) {
    }
    var currentContext = LocalContext.current
    Column(
        modifier = Modifier.fillMaxSize(),
        verticalArrangement = Arrangement.Bottom,
        horizontalAlignment = Alignment.CenterHorizontally,
    ){
        Button(onClick = {captureImage(cameraClient, currentContext)}) {
            Text("Take Picture")
        }
    }
}
```
The log output shows it doesn't change from null
```
2022-09-30 09:49:52.384 14054-14054/com.example.camerauvctest I/testing: camView is null
2022-09-30 09:49:52.767 14054-14054/com.example.camerauvctest I/testing: camView is null
```

I tried to set a CameraView, but it defaults to null even when I setAspectRatio. I can take 2k res photo if I set `cameraClient.openCamera(camView)` but then the preview goes black.

jiangdongguo, 2022-09-30:
First, you should try 2592x1944 by using demo's updating resolution function and make sure it's ok. Then, if can't please send me your logs otherwise check your own code logical

shanerodrigues, 2022-09-30:
I tried using `cameraClient.updateResolution()` but every time after permission is given the app crashes.
Here's the log https://pastebin.com/XGbzTSRK
The error I get is
```
2022-09-30 13:30:16.508 10662-10715/? E/GLConsumer: [SurfaceTexture-0-10662-1] checkAndUpdateEglState: invalid current EGLDisplay
2022-09-30 13:30:16.514 10662-10735/? W/libUVCCamera: [10735*UVCPreview.cpp:128:get_frame]:allocate new frame
2022-09-30 13:30:16.523 10662-10715/? E/AndroidRuntime: FATAL EXCEPTION: gl_render
    Process: com.example.camerauvctest, PID: 10662
    java.lang.IllegalStateException: Unable to update texture contents (see logcat for details)
        at android.graphics.SurfaceTexture.nativeUpdateTexImage(Native Method)
        at android.graphics.SurfaceTexture.updateTexImage(SurfaceTexture.java:243)
        at com.jiangdg.ausbc.render.RenderManager.handleMessage(RenderManager.kt:162)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
```

Luke-Tang, 2022-09-30:
try these
```kotlin   
 setEnableGLES(false)
 setRawImage(true)
```

shanerodrigues, 2022-10-04:
@Luke-Tang the app crashes after granting permission to use the camera.

shanerodrigues, 2022-10-04:
> try these
> 
> ```kotlin
>  setEnableGLES(false)    // disable opengl
>  setRawImage(true)      // capture image from the origin preview frame 
> ```

After doing more testing, this made it work. I get the preview and correct resolution photo. Thanks @Luke-Tang!

closed 2022-10-04 by shanerodrigues