jiangdongguo/AndroidUSBCamera #472: 旋转屏幕后黑屏不显示  -99

biqi12321, 2022-10-01:
2022-10-01 18:38:00.377 14118-14191/com.example.usbcamera_test D/UsbDeviceConnectionJNI: close
2022-10-01 18:38:00.378 14118-14191/com.example.usbcamera_test I/libusb/core: [14191*core.c:2085:libusb_init2]:libusb v1.0.19.10903
2022-10-01 18:38:00.378 14118-14191/com.example.usbcamera_test I/libusb/usbfs: [14191*android_usbfs.c:504:op_init2]:start up hotplug event handler
2022-10-01 18:38:00.378 14118-14191/com.example.usbcamera_test I/libusb/usbfs: [14191*android_usbfs.c:512:op_init2]:call android_scan_devices
2022-10-01 18:38:00.390 14118-14192/com.example.usbcamera_test D/UsbDeviceConnectionJNI: close
2022-10-01 18:38:00.390 14118-14192/com.example.usbcamera_test I/libusb/core: [14192*core.c:2085:libusb_init2]:libusb v1.0.19.10903
2022-10-01 18:38:00.390 14118-14192/com.example.usbcamera_test I/libusb/usbfs: [14192*android_usbfs.c:512:op_init2]:call android_scan_devices
2022-10-01 18:38:00.395 14118-14192/com.example.usbcamera_test E/libusb/usbfs: [14192*android_usbfs.c:1776:release_interface]:release interface failed, error -1 errno 22
2022-10-01 18:38:00.395 14118-14192/com.example.usbcamera_test E/UVCCamera: [14192*UVCCamera.cpp:172:connect]:could not open camera:err=-99
2022-10-01 18:38:00.396 14118-14192/com.example.usbcamera_test E/CameraUvc:  preview failed, err = open failed:result=-99----->id_camera=-5476376666539599232;venderId=7119;productId=11444;fileDescriptor=121;busNum=1;devAddr=2;usbfs=/dev/bus/usb
    Exception：调用nativeConnect返回值：-99
    java.lang.UnsupportedOperationException: open failed:result=-99----->id_camera=-5476376666539599232;venderId=7119;productId=11444;fileDescriptor=121;busNum=1;devAddr=2;usbfs=/dev/bus/usb
    Exception：调用nativeConnect返回值：-99
        at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:215)
        at com.jiangdg.ausbc.camera.CameraUvcStrategy.createCamera(CameraUvcStrategy.kt:149)
        at com.jiangdg.ausbc.camera.CameraUvcStrategy.startPreviewInternal(CameraUvcStrategy.kt:121)
        at com.jiangdg.ausbc.camera.ICameraStrategy.handleMessage(ICameraStrategy.kt:103)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:236)
        at android.os.HandlerThread.run(HandlerThread.java:67)
2022-10-01 18:38:00.396 14118-14118/com.example.usbcamera_test E/EventBus.CAMERA:: -1
2022-10-01 18:38:00.396 14118-14118/com.example.usbcamera_test E/EventBus.CAMERA:: -1
2022-10-01 18:38:00.397 14118-14118/com.example.usbcamera_test E/EventBus.CAMERA:: 2
2022-10-01 18:38:00.397 14118-14118/com.example.usbcamera_test E/EventBus.CAMERA:: 2
2022-10-01 18:38:00.441 14118-14191/com.example.usbcamera_test I/CameraUvc: getAllPreviewSizes = [Size(2592x1944@ 0.0,type:6,frame:0,index:0,[]), Size(640x480@ 0.0,type:6,frame:0,index:1,[]), Size(2560x1440@ 0.0,type:6,frame:0,index:2,[]), Size(1280x720@ 0.0,type:6,frame:0,index:3,[]), Size(1920x1080@ 0.0,type:6,frame:0,index:4,[]), Size(800x600@ 0.0,type:6,frame:0,index:5,[]), Size(1600x1200@ 0.0,type:6,frame:0,index:6,[]), Size(960x720@ 0.0,type:6,frame:0,index:7,[]), Size(1280x960@ 0.0,type:6,frame:0,index:8,[])]
2022-10-01 18:38:00.485 14118-14191/com.example.usbcamera_test I/libUVCCamera: [14191*UVCPreview.cpp:285:callbackPixelFormatChanged]:PIXEL_FORMAT_YUV20SP:
2022-10-01 18:38:00.486 14118-14191/com.example.usbcamera_test I/CameraUvc:  createCamera success! request = CameraRequest(previewWidth=1280, previewHeight=720, cameraId='1002', isFrontCamera=false, isContinuousAFModel=true, isContinuousAEModel=true)

jiangdongguo, 2022-10-09:
摄像头释放有问题，你应该处理下让Activity屏幕旋转不重新创建

       <activity
            android:name=".activity.HomeActivity"
            android:configChanges="keyboardHidden|orientation|screenLayout|screenSize|smallestScreenSize"
            android:launchMode="singleInstance"
            android:theme="@style/HomeActivityStyle"
            android:screenOrientation="portrait" />

biqi12321, 2022-10-09:
嗯，明白后面也这么处理的，谢谢你的帮助

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2022年10月09日 09:49 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.******@***.***> |
| 主题 | Re: [jiangdongguo/AndroidUSBCamera] 旋转屏幕后黑屏不显示  -99 (Issue #472) |

摄像头释放有问题，你应该处理下让Activity屏幕旋转不重新创建

   <activity

        android:name=".activity.HomeActivity"

        android:configChanges="keyboardHidden|orientation|screenLayout|screenSize|smallestScreenSize"

        android:launchMode="singleInstance"

        ***@***.***/HomeActivityStyle"

        android:screenOrientation="portrait" />

jiangdongguo, 2022-10-10:
没事~

closed 2022-10-10 by jiangdongguo