jiangdongguo/AndroidUSBCamera #473: Dependancy issues after adding to project

eetteri, 2022-10-05:
I'm trying to incorporate the lib into a project. After I add
```
implementation 'com.github.jiangdongguo:AndroidUSBCamera:3.2.8' 
```
to build.gradle I start getting following dependancy errors
```
Failed to resolve: com.gyf.immersionbar:immersionbar:3.0.0
Failed to resolve: com.zlc.glide:webpdecoder:1.6.4.9.0
```

Is this my newbieness with Android Studio or what could cause this?

jiangdongguo, 2022-10-09:
Please update to 3.2.9

eetteri, 2022-10-09:
Updated to 3.2.9 and the end result is same. Below is the whole list I have in use.

```
    implementation 'com.github.girish3.Vlog:library:v0.4'
    implementation 'com.github.jiangdongguo:AndroidUSBCamera:3.2.9'
    implementation 'androidx.appcompat:appcompat:1.5.1'
    implementation 'com.google.android.material:material:1.6.1'
    implementation 'androidx.constraintlayout:constraintlayout:2.1.4'
    testImplementation 'junit:junit:4.13.2'
    androidTestImplementation 'androidx.test.ext:junit:1.1.3'
    androidTestImplementation 'androidx.test.espresso:espresso-core:3.4.0'
    implementation 'io.agora.rtc:full-sdk:3.7.1'
    implementation "androidx.core:core-ktx:+"
```

If AndroidUSBCamera is enabled I still get the errors below and when commented out the errors disappear.

```
Failed to resolve: com.gyf.immersionbar:immersionbar:3.0.0
Failed to resolve: com.zlc.glide:webpdecoder:1.6.4.9.0
````

jiangdongguo, 2022-10-10:
Add those to your app gradle file:
   
 implementation 'com.gyf.immersionbar:immersionbar:3.0.0'
 implementation "com.zlc.glide:webpdecoder:1.6.4.9.0"

jiangdongguo, 2022-10-11:
please update to 3.2.10 and have a try.

eetteri, 2022-10-11:
Bumped the version up to 3.2.10, still the result below.

[image]

jiangdongguo, 2022-10-12:
Have your project used Glide or Immersionbar?It seems could not download it, may you can try remove it.

eetteri, 2022-10-12:
I'm not using Glide or Immersionbar in my project. The dependancies and errors appear only if I add the AndroidUSBCamera to my project. If I comment out AndroidUSBCamera project builds and runs just fine, screenshots below. 

Without AndroidUSBCamera
[image]

With AndroidUSBCamera
[image]

jiangdongguo, 2022-10-13:
implementation 'com.github.jiangdongguo.AndroidUSBCamera:libausbc:3.2.10'

NOT：

implementation 'com.github.jiangdongguo.AndroidUSBCamera:3.2.10'

eetteri, 2022-10-13:
@jiangdongguo thank you for your patience and help. All works great again.

closed 2022-10-13 by eetteri