jiangdongguo/AndroidUSBCamera #474: how to change frame format to mjpeg on version 3.2.8?

rizkyhhh, 2022-10-06:
hello I would like to thank you for the library, its really help me a lot in my research. I wanna ask how to change the frame format from YUV to MJPEG on the 3.2.8 version, I tried to change it manually from CameraUvcStrategy.kt and on UVCCamera.kt directly but it still running on YUV format (I am assuming that because its run on 5-10 fps on 1980x1080 resolution). and also I cant implement mCameraHelper?.setdefaultformat(FRAME_FORMAT_MJPEG) because its been removed in newer version I guess?
[image]

androidscx, 2022-10-08:
It seems that the chip decides whether to output data in YUV format or MJPEG format?

rizkyhhh, 2022-10-09:
i have tried to commented out the option that the app will choose YUV format, but the app still run on low fps at 1980x1080

jiangdongguo, 2022-10-09:
Hi, 3.0+ default using MHPEG format, If your device do not support it, it will use YUV. So make sure your device support the MJPEG format first!

closed 2022-10-10 by jiangdongguo