jiangdongguo/AndroidUSBCamera #476: No implementation found for long com.jiangdg.uvc.UVCCamera.nativeCreate()

timzaak, 2022-10-10:
```
CameraClient.newBuilder(this)
                    .setEnableGLES(true)
                    .setCameraStrategy(new CameraUvcStrategy(this))
                    .setCameraRequest(new CameraRequest.Builder()
                            .setFrontCamera(false)
                            .setPreviewHeight(480)
                            .setPreviewHeight(640)
                            .create()
                    )
                    .openDebug(true)
                    .build();
```

```
 cameraClient.openCamera(null, true);
```

Then it would occure error:
```
2022-10-10 18:05:48.133 20756-20827/? E/angzv.uvccamer: No implementation found for long com.jiangdg.uvc.UVCCamera.nativeCreate() (tried Java_com_jiangdg_uvc_UVCCamera_nativeCreate and Java_com_jiangdg_uvc_UVCCamera_nativeCreate__)
2022-10-10 18:05:48.140 20756-20827/? E/AndroidRuntime: FATAL EXCEPTION: camera_manager
    Process: com.angzv.uvccamera, PID: 20756
    java.lang.UnsatisfiedLinkError: No implementation found for long com.jiangdg.uvc.UVCCamera.nativeCreate() (tried Java_com_jiangdg_uvc_UVCCamera_nativeCreate and Java_com_jiangdg_uvc_UVCCamera_nativeCreate__)
        at com.jiangdg.uvc.UVCCamera.nativeCreate(Native Method)
        at com.jiangdg.uvc.UVCCamera.<init>(UVCCamera.java:179)
        at com.jiangdg.ausbc.camera.CameraUvcStrategy.createCamera(CameraUvcStrategy.kt:128)
        at com.jiangdg.ausbc.camera.CameraUvcStrategy.startPreviewInternal(CameraUvcStrategy.kt:110)
        at com.jiangdg.ausbc.camera.ICameraStrategy.handleMessage(ICameraStrategy.kt:97)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:236)
        at android.os.HandlerThread.run(HandlerThread.java:67)
```

If ` .setEnableGLES(false)` , this only output log, then do nothing.
```
2022-10-10 18:10:02.916 20983-20983/com.angzv.uvccamera I/CameraClient: captureImage...
```

If it's now possible to caputure picture without preview?

jiangdongguo, 2022-10-11:
Which version？Please update to 3.2.10 and have a try.By the way libuvc's project structure has changed!

timzaak, 2022-10-11:
I tried with the latest source code.
change https://github.com/jiangdongguo/AndroidUSBCamera/blob/8270a4effbc7219525367dbe3af1c5be05e3b3b3/app/src/main/java/com/jiangdg/demo/DemoMultiCameraFragment.kt#L58
 =>  `cam.openCamera(null, getCameraRequest())`

Then App would crash.

I wannt use  MultipleCamera without android UI (in Flutter), If it's possible?

想在 不绘制UI 的前提下进行拍照，是否可行？(这样能简单集成到Flutter 中去)

jiangdongguo, 2022-10-12:
You can only use CameraClient or MultiCameraClient to do something instead of extending CameraFragment or CameraActivity.Of course, passing null is supported and let me check the code.You can also paste the crash information, so I can find the error quickly.

timzaak, 2022-10-12:
There's no log output, I have forked and push a simple example. [timzaak/AndroidUSBCamera](https://github.com/timzaak/AndroidUSBCamera)
commit is: https://github.com/timzaak/AndroidUSBCamera/commit/089f108c6adbe303738a2a2f91989fe3dc076db5
You can clone and have a try.

点击拍照，没有任何日志出现。

我提交了 example 代码，另外，由于`import com.jiangdg.usb.USBMonitor.UsbControlBlock` 会导致 `Unresolved reference`  错误，故用 UsbManager 来获得 UsbDevice.

jiangdongguo, 2022-10-13:
OK, I will check it. Ths.

closed 2022-10-25 by timzaak