jiangdongguo/AndroidUSBCamera #483: Change video size

ShafiqSadat, 2022-10-27:
the video size is not same as preview 
in preview size is same as source but when I record video size is different! for example : 

https://imgur.com/a/KdsiH2A

I'm using your example!

sorry for bad English!

jiangdongguo, 2022-10-28:
Hi, try like this:

`private fun initEncodeProcessor() {
        releaseEncodeProcessor()
        /*val  encodeWidth = if (isEnableGLEs) {
            mRequest!!.previewHeight
        } else {
            mRequest!!.previewWidth
        }
        val encodeHeight = if (isEnableGLEs) {
            mRequest!!.previewWidth
        } else {
            mRequest!!.previewHeight
        }*/
        val encodeWidth =  mRequest!!.previewWidth
        val encodeHeight = mRequest!!.previewHeight
        mAudioProcess = AACEncodeProcessor()
        mVideoProcess = H264EncodeProcessor(encodeWidth, encodeHeight, /*isEnableGLEs*/true)
    }`

If ok. tell me.

ShafiqSadat, 2022-10-29:
where should put this code?
which class?

jiangdongguo, 2022-10-31:
H264Encode

rizkyhhh, 2022-11-03:
> where should put this code? which class?

you can find it on CameraClient.kt and search for initEncodeProcessor() function