jiangdongguo/AndroidUSBCamera #485: App doesn't work on my blackview smartphone

JulesD29, 2022-10-28:
Hello everyone,
I don’t know if this has ever happened to anyone, but the @jiangdongguo library doesn’t work with my Blackview OSCAL S60 (Android 11) and my Blackview 5200 (Android 12). However, on my phone Xiaomi Redmi note 11 Pro 5G (Android 11) the app works perfectly.

I gave all the permissions requested but nothing appears on the screen, there is no preview that appears for BlackView phones.

If you have possible solutions I am a taker! Thank you in advance

Jules

jiangdongguo, 2022-10-28:
Hi,
Have you phone show usb permission dialog? Please show your logs at Android/data/com.jiangdg.ausbc.

GTX-GSO, 2022-11-04:
Blackview BL8800也是一样黑屏，问了多科那边，得到的明确答复是给钱也不做适配，目前尝试了一圈能正常工作的只有这个[https://www.shenyaocn.com/](url)，同样基于saki4510t的项目二次开发，可以在MTK天玑系列设备上运行，但是作者在play上架收费商用了，不开放源码且对so库进行了加密，目前推测可能是由于libuvc、libusb这两个库年代久远，但是水平有限，真的改不动

Clamorae, 2022-11-04:
Hi,

yes the phone is asking for usb permission, it recognize the camera but show only a black screen, there is the same problem on ulefone.

[log.txt](https://github.com/jiangdongguo/AndroidUSBCamera/files/9936370/log.txt)

GTX-GSO, 2022-11-06:
目前排查下来，这些预览黑屏的手机ROM都缺少UVC驱动，插入设备并确认授权后被没有video节点新增，只在bus/下新增了usb设备

Clamorae, 2022-11-07:
Do you know any way to install UVC driver on android 11/12 phone ?

Clamorae, 2022-11-07:
After some trym we figured out that the app work on blackview 5100 and on cubot pocket which doesn't have UVC driver.
in other hands, Blackview 5200, oscal S60 and ulefone armor X6 don't have UVC driver but the app isn't working on them.
Maybe adding UVC driver on the phone can work, do you have any idea on how to install it?

GTX-GSO, 2022-11-08:
> After some try, we figured out that the app work on blackview 5100 and on cubot pocket which doesn't have UVC driver. In other hands, Blackview 5200, oscal S60 and ulefone armor X6 don't have UVC driver but the app isn't working on them. Maybe adding UVC driver on the phone can work, do you have any idea on how to install it on android 11/12 phone?

emm...I'm afraid that u can't just Install a Linux Driver by apk , It needs to be place in System , to flash a ROM that contains UVC driver will do that, and we are contacting BlackView to provide these ROMs ,but I can't share because its for bussniess use only 
plus even with uvc driver this lib will still not working , It connect camera using USB protocol not V4L2 framework , with uvc driver, plug in a camera will provied u a video* endpoint in /dev/, but still need to change selinux to premissive and get read write permission on this endpoint. so use this way only if you are develop System App or install on rooted phones

Clamorae, 2022-11-08:
Is there any workaround ? Is there something else we can do to use this app on Blackview smartphone, or is this a non-go?

GTX-GSO, 2022-11-09:
> Is there any workaround ? Is there something else we can do to use this app on Blackview smartphone, or is this a non-go?

try update libyuv libusb version of this app from 2014 to 2022 , I'm already given up this way ，to much complicated than I expected, I also tried decompile this app https://play.google.com/store/apps/details?id=com.shenyaocn.android.usbcamera ,but cant get its SOs loaded on my project.