jiangdongguo/AndroidUSBCamera #490: 请问推流到流媒体服务器应该做些什么?

xxxxxxbin6, 2022-11-08:


xxxxxxbin6, 2022-11-08:
我想要将usb摄像头的流推到服务器,在哪能设置流地址

xxxxxxbin6, 2022-11-08:
我看有一个startPush方法

jiangdongguo, 2022-11-11:
目前没实现推流。自己接推流模块，然后拿H264和AAC推送

xxxxxxbin6, 2022-11-11:
目前实现了推流,这个想不预览拿到实时流可行吗,我看直接继承CameraFragment才能有addPreviewDataCallBack回调

xxxxxxbin6, 2022-11-14:
我在随便一个activity(不是继承CameraActivity)中创建CameraClient实例,拿不到addPreviewDataCallback回调数据吗

xxxxxxbin6, 2022-11-14:
[image]

xxxxxxbin6, 2022-11-14:
以上是创建client实例代码

lxfwelldone, 2022-12-06:
你好，我刚好也需要实现同样的功能。请问如何实现？ 怎么写才能获取视频流和音频流？

class TestCameraFragment: CameraFragment(), IEncodeDataCallBack {

    companion object {
        const val TAG: String = "TestCameraFragment"
    }

    override fun getCameraView(): IAspectRatio {
        return AspectRatioTextureView(requireContext())
    }

    override fun getCameraViewContainer(): ViewGroup? {
       return getRootView()?.findViewById(R.id.camera_view_container)
    }

    override fun getRootView(inflater: LayoutInflater, container: ViewGroup?): View? {
        return inflater.inflate(R.layout.fragment_camera_test, null)
    }

    override fun getCameraClient(): CameraClient {
        return CameraClient.newBuilder(requireContext())
            .setEnableGLES(true)
            .setRawImage(true)
            .setCameraStrategy(CameraUvcStrategy(requireContext()))
            .setCameraRequest(getCameraRequest())
            .openDebug(true)
            .build()
    }
    private fun getCameraRequest(): CameraRequest {
        return CameraRequest.Builder()
            .setFrontCamera(false)
            .setPreviewWidth(1920)
            .setPreviewHeight(1080)
            .create()
    }

    override fun initData() {
        super.initData()

        getCameraClient().addEncodeDataCallBack(this)
        getCameraClient().startPush()
    }

    override fun onEncodeData(data: ByteArray?, size: Int, type: IEncodeDataCallBack.DataType) {
        Log.d(TAG, "onEncodeData: data=${data.toString()} size=$size type=$type")
    }

jiangdongguo, 2023-02-11:
更新到3.3.2， YUV/RGBA/PCM/H.264/AAC都可以拿得到，CameraFragment等基类都提供了对应的接口

gatspy, 2024-08-30:
能提供个推送的样例吗